ros-simulation/gazebo_ros_pkgs #838: [ros2] - /gazebo/set_model_state service isn't showing up

mkhansen-intel, 2018-11-14:
I'm trying to call the set_model_state service, but I don't see it listed when I do a service list, and when I try to call it I get `waiting for service to become available...`

```
mkhansen@mkhansen-desk:~$ ros2 service list | grep gazebo
/gazebo/describe_parameters
/gazebo/get_parameter_types
/gazebo/get_parameters
/gazebo/list_parameters
/gazebo/set_parameters
/gazebo/set_parameters_atomically

```
Is this supposed to be available in ROS2? I'm running gazebo with this command:
```
gazebo --verbose -s libgazebo_ros_init.so /home/mkhansen/ros2_dev/turtlebot3_ws/src/turtlebot3_simulations/turtlebot3_gazebo/worlds/turtlebot3_ros2_demo.world
```

mkhansen-intel, 2018-11-14:
I should have mentioned:
Ubuntu 18.04
Gazebo 9

chapulina, 2018-11-14:
No, that service hasn't been ported to ROS2 yet, see #779. If that's blocking you, I can try to make some time to port it, it shouldn't be too hard.

chapulina, 2018-11-14:
I've assigned the model states services and topics to myself, I'll tackle it asap.

mkhansen-intel, 2018-11-14:
Thanks @chapulina

mhpanah, 2019-02-14:
> I've assigned the model states services and topics to myself, I'll tackle it asap.

@chapulina Has `set_model_state` service ported yet?

chapulina, 2019-02-14:
It's currently in a pull request: https://github.com/ros-simulation/gazebo_ros_pkgs/pull/839

Given the time it's been up for review, I'm inclined to merge it if I get a positive review from someone in the community.

nzlz, 2019-02-14:
We have been using a [merged gazebo_ros_pkgs](https://github.com/AcutronicRobotics/gazebo_ros_pkgs/tree/ros2_gym-gazebo) with ros2_time_cmds, ros2_state and ros2_properties (my PR) for the last 20 days, intensively on a daily basis. Mainly for the upcoming version of gym_gazebo. No errors so far, I'd say ros2_state is ready to merge and others too.

chapulina, 2019-02-14:
Thanks for the feedback, @nzlz ! I'll rebase all the PRs and run CI one more time tomorrow.  If CI comes back clean, I'll merge them and prepare a new release.

chapulina, 2019-02-14:
#839 merged

closed 2019-02-14 by chapulina

i1Cps, 2023-05-07:
Hello, Im running ros2 on Foxy, and the service set_model_state is still not showing. I'm seeing exactly what the OP intially posted. Has the service been ported yet? 

OS: Ubuntu 20.04
Ros2: Foxy
Gazebo: version 11.11.0

i1Cps, 2023-06-26:
> Hello, Im running ros2 on Foxy, and the service set_model_state is still not showing. I'm seeing exactly what the OP intially posted. Has the service been ported yet?
> 
> OS: Ubuntu 20.04 Ros2: Foxy Gazebo: version 11.11.0

Sorry for the confusion but the reason the service wasn't showing up for me was because I hadn't added the plugin to the world file. Once I added the plugin for the service at the top of the world like this:
`<plugin name='set_entity_state_plugin' filename='libset_entity_state.so'>`
`     <robot_name>my_robot</robot_name>`
` </plugin>`

The service showed up after the `ros2 service list` command

NamTruongTran, 2023-09-21:
> Sorry for the confusion but the reason the service wasn't showing up for me was because I hadn't added the plugin to the world file. Once I added the plugin for the service at the top of the world [...]

Which world file exactly?

i1Cps, 2023-09-21:
Hello, I have this, at the start of my world file:
```
<world name='default'> 
    <plugin name='set_entity_state_plugin' filename='libset_entity_state.so'>
      <robot_name>my_robot</robot_name>
    </plugin>
    
    <plugin name='gazebo_ros_state' filename='libgazebo_ros_state.so'>
      <ros>
        <argument>model_states:=model_states_demo</argument>
      </ros>
      <update_rate>1.0</update_rate>
    </plugin>
   ```
    
    

> @i1Cps do you had to change anything else to make it run? I have added this snippet just below my ` <world name='orl'>` tag. However, the service is still not showing up.