ros-simulation/gazebo_ros_pkgs #840: [ROS2] tf publisher stamp

routiful, 2018-11-19:
I have tried to make map using Cartographer but I found that the time is different between two nodes.

I ran a tf2_echo to show that the tf is working well. Left terminal shows tf2_echo between map frame and odom frame. Right terminal shows tf2_echo between odom frame and base_footprint.
You can find that the time has quite different.

[image]

Because Cartographer using chrono or ROS time however the diff_drive in ROS2 package publishes odometry tf with gazebo common time.

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/73a12b7c9bbfebd9a0030bc4f2f394f64228c05e/gazebo_plugins/src/gazebo_ros_diff_drive.cpp#L492

How can I sync this two nodes?

chapulina, 2018-11-20:
You need to make sure that:

* You launch Gazebo with the `gazebo_ros_init` plugin
* Set `use_sim_time` on your Cartographer nodes

See the migration guide [here](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-ROS-Clocks-and-sim-time).

chapulina, 2018-11-20:
Let me know if that works for you and then we can close this issue.

routiful, 2018-11-21:
Great! It works :)

closed 2018-11-21 by routiful