ros-simulation/gazebo_ros_pkgs #843: p3d sensor wrong behavior when using fixed joints

reinzor, 2018-11-27:
When I specify a `bodyName` that is attached to a another link with a fixed joint, the link name resolves to the parent: https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_plugins/src/gazebo_ros_p3d.cpp#L68

reinzor, 2018-11-27:
I workaround would be adding an additional parameter that overwrites the `child_frame_id` in the odom message and use the `xyzOffset` parameter. What do you think?

reinzor, 2018-11-27:
Suggested fix: https://github.com/ros-simulation/gazebo_ros_pkgs/pull/844

KitKat7, 2018-11-28:
I met the same problem, but after I add the following snippets to the `urdf` file, this problem disappeared.
```xml
<gazebo reference="some_joint">
  <disableFixedJointLumping>true</disableFixedJointLumping>
</gazebo>
```
However, I think the p3d in current version ef4da48492c60f6a06f2fb2e057c201314bf9f03 might have some bugs (especially the relative transformations)...

reinzor, 2018-11-28:
Thanks, this seems to work !

closed 2018-11-29 by reinzor