ros-simulation/gazebo_ros_pkgs #851: [ros2] Plugins failing to load

chapulina, 2018-12-09:
I just tried the Crystal debs and no plugins from `gazebo_plugins` are loading properly, but plugins from `gazebo_ros` like `libgazebo_ros_factory.so` are loading.

Then I also built `gazebo_ros_pkgs` from source against Crystal debs and got the same result. The errors look like this:

`[Err] [Model.cc:1129] Exception occured in the Load function of plugin with name[gazebo_ros_force] and filename[libgazebo_ros_force.so]. This plugin will not run.  `

This was working a couple of days ago, so it looks like some upstream change may have affected these plugins. I haven't tried debugging further, but maybe one of our tests gives a better hint of what exactly is failing.

chapulina, 2018-12-10:
More detailed error from running `test_plugins`:

    terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidArgument'
      what():  failed to create interrupt guard condition: context argument is null, at /tmp/binarydeb/ros-crystal-rcl-0.6.1/src/rcl/guard_condition.c:65

This may be related to the changes in init and context on https://github.com/ros2/rclcpp/pull/587

chapulina, 2018-12-10:
Fix in a103bfb3101bb0ba208a9245345d2339ff6ba589, which is now part of https://github.com/ros-simulation/gazebo_ros_pkgs/pull/849

closed 2018-12-11 by chapulina

utsavrai, 2022-01-18:
Hi I am using foxy branch of this repository and facing similar issue, I have included the gazebo_ros_plugin for simulating universal robot ur5e. I was able to spawn the robot, but when loading the plugin, the gzserver throws `[gzserver-10] Segmentation fault (core dumped)` and ` [Err] [Model.cc:1157] Exception occured in the Load function of plugin with name[gazebo_ros2_control] and filename[libgazebo_ros2_control.so]. This plugin will not run.`

```
ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.43.8 use_fake_hardware:=true launch_rviz:=false debug:=true verbose:=true
[INFO] [launch]: All log files can be found below /home/utsav/.ros/log/2022-01-18-13-35-08-235068-01hw1321985-6889
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [6895]
[INFO] [ros2_control_node-1]: process started with pid [6893]
[INFO] [robot_state_publisher-3]: process started with pid [6897]
[INFO] [spawner.py-4]: process started with pid [6899]
[INFO] [spawner.py-5]: process started with pid [6901]
[INFO] [spawner.py-6]: process started with pid [6903]
[INFO] [spawner.py-7]: process started with pid [6905]
[INFO] [spawner.py-8]: process started with pid [6907]
[INFO] [spawn_ur_bot.py-9]: process started with pid [6909]
[INFO] [gzserver-10]: process started with pid [6911]
[INFO] [gzclient --verbose  -11]: process started with pid [6913]
[robot_state_publisher-3] Parsing robot urdf xml string.
[robot_state_publisher-3] Link base_link had 2 children
[robot_state_publisher-3] Link base had 0 children
[robot_state_publisher-3] Link base_link_inertia had 1 children
[robot_state_publisher-3] Link shoulder_link had 1 children
[robot_state_publisher-3] Link upper_arm_link had 1 children
[robot_state_publisher-3] Link forearm_link had 1 children
[robot_state_publisher-3] Link wrist_1_link had 1 children
[robot_state_publisher-3] Link wrist_2_link had 1 children
[robot_state_publisher-3] Link wrist_3_link had 1 children
[robot_state_publisher-3] Link flange had 1 children
[robot_state_publisher-3] Link tool0 had 0 children
[robot_state_publisher-3] [INFO] [1642493109.395013233] [robot_state_publisher]: got segment base
[robot_state_publisher-3] [INFO] [1642493109.395117243] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1642493109.395129396] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-3] [INFO] [1642493109.395137112] [robot_state_publisher]: got segment flange
[robot_state_publisher-3] [INFO] [1642493109.395144160] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-3] [INFO] [1642493109.395151169] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-3] [INFO] [1642493109.395157894] [robot_state_publisher]: got segment tool0
[robot_state_publisher-3] [INFO] [1642493109.395164336] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-3] [INFO] [1642493109.395171151] [robot_state_publisher]: got segment world
[robot_state_publisher-3] [INFO] [1642493109.395178226] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-3] [INFO] [1642493109.395185194] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-3] [INFO] [1642493109.395192077] [robot_state_publisher]: got segment wrist_3_link
[ros2_control_node-1] [INFO] [1642493109.419928136] [controller_manager]: update rate is 600 Hz
[spawner.py-7] [INFO] [1642493109.577527329] [spawner_force_torque_sensor_broadcaster]: Waiting for /controller_manager services
[ros2_control_node-1] [INFO] [1642493109.582783863] [controller_manager]: Loading controller 'joint_state_broadcaster'
[ros2_control_node-1] [INFO] [1642493109.591043347] [controller_manager]: Loading controller 'io_and_status_controller'
[spawner.py-4] [INFO] [1642493109.599968555] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-1] [INFO] [1642493109.601338475] [controller_manager]: Loading controller 'speed_scaling_state_broadcaster'
[spawner.py-5] [INFO] [1642493109.614758985] [spawner_io_and_status_controller]: Loaded io_and_status_controller
[ros2_control_node-1] [INFO] [1642493109.615833372] [controller_manager]: Configuring controller 'io_and_status_controller'
[spawner.py-6] [INFO] [1642493109.620411223] [spawner_speed_scaling_state_broadcaster]: Loaded speed_scaling_state_broadcaster
[ros2_control_node-1] [INFO] [1642493109.623917952] [controller_manager]: Configuring controller 'speed_scaling_state_broadcaster'
[ros2_control_node-1] [INFO] [1642493109.623962862] [speed_scaling_state_broadcaster]: Publisher rate set to : 100.0 Hz
[spawner.py-5] [INFO] [1642493109.624532453] [spawner_io_and_status_controller]: Configured and started io_and_status_controller
[spawner.py-6] [INFO] [1642493109.629626698] [spawner_speed_scaling_state_broadcaster]: Configured and started speed_scaling_state_broadcaster
[INFO] [spawner.py-5]: process has finished cleanly [pid 6901]
[INFO] [spawner.py-6]: process has finished cleanly [pid 6903]
[ros2_control_node-1] [INFO] [1642493109.681695403] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ros2_control_node-1] [INFO] [1642493109.684450936] [controller_manager]: Loading controller 'joint_trajectory_controller'
[spawner.py-4] [INFO] [1642493109.691665754] [spawner_joint_state_broadcaster]: Configured and started joint_state_broadcaster
[... cut ...]
[gzclient --verbose  -11] [Err] [GLWidget.cc:900] Unable to connect to a running Gazebo master.
[gzclient --verbose  -11] [Wrn] [Publisher.cc:136] Queue limit reached for topic /gazebo/default/user_camera/pose, deleting message. This warning is printed only once.
[gzclient --verbose  -11] [Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
[INFO] [gzclient --verbose  -11]: process has finished cleanly [pid 6913]
```
I am unable to figure the issue here, please let me know your thoughts on it.