ros-simulation/gazebo_ros_pkgs #857: error: ‘is_initialized’ is not a member of ‘rclcpp’

RDaneelOlivav, 2018-12-13:
When compiling the ros2 branch ( through the command `colcon build`), for ROS2 bouncy installed from debs not sourcecode, it gives the following error:

In file included from /home/ubuntu/ros2_ws/src/gazebo_ros_pkgs/gazebo_ros/src/node.cpp:15:0:
/home/ubuntu/ros2_ws/src/gazebo_ros_pkgs/gazebo_ros/include/gazebo_ros/node.hpp: In static member function ‘static gazebo_ros::Node::SharedPtr gazebo_ros::Node::CreateWithArgs(Args&& ...)’:
/home/ubuntu/ros2_ws/src/gazebo_ros_pkgs/gazebo_ros/include/gazebo_ros/node.hpp:139:15: error: ‘is_initialized’ is not a member of ‘rclcpp’
   if (rclcpp::is_initialized()) {
               ^~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/gazebo_ros_node.dir/src/node.cpp.o] Error 1
make[1]: *** [CMakeFiles/gazebo_ros_node.dir/all] Error 2
make: *** [all] Error 2

Any ideas on what it could be happening here?

mjcarroll, 2018-12-13:
I believe that you will need to either use the (still-testing) Crystal debs, or build from source.  The Bouncy debs will not work with the latest state of the `ros2` branch.

chapulina, 2018-12-13:
All the repositories in [ros2.repos](https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos)'s master branch, plus a couple that you can see [here](https://bitbucket.org/snippets/chapulina/geRKyA).

Or you can use Crystal pre-release debs, or wait for the official release in a few days.

RDaneelOlivav, 2018-12-14:
Thanks, @chapulina  ;). We'll look into it then :). Excited to play with RO2 and gazebo :)

chapulina, 2018-12-14:
Cool, feel free to ticket any issues you find, or ask for help at http://answers.gazebosim.org . Have fun!

closed 2018-12-14 by chapulina