ros-simulation/gazebo_ros_pkgs #863: [ros2] Reset Simulation/World

LuisYordano, 2018-12-29:
Hi **gazebo_ros_pkgs ROS2 Team**,

In ROS1 there are two services:

1. rosservice call /gazebo/reset_simulation "{}" 
2. rosservice call /gazebo/reset_world "{}"

In ROS2 there are not these services: **/gazebo/reset_simulation** and **/gazebo/reset_world**

Could you tell me what the equivalent command is in ROS2?

Thanks in advance for your answer

nzlz, 2018-12-31:
I guess this has not been ported yet, but you could make use of the spawn/delete_entity services meanwhile: https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-Spawn-and-delete

If you want reset only for model-resetting purposes: _delete + spawn = reset_

I will be testing this right now so I will post again if I find some interesting tips. It would be a good idea to pause the simulation while this process occurs. I see this is now handled in via `gazebo_ros_init` ([link](https://github.com/ros-simulation/gazebo_ros_pkgs/issues/779)).

LuisYordano, 2018-12-31:
Hi Nestor,
Thanks for your answer.
If you find a tip, posting it will be helpful until the migration is made.

nzlz, 2018-12-31:
# Can't re-spawn a complex robot.
My problems come when I try to spawn the [MARA](https://github.com/AcutronicRobotics/MARA) back again. First, I delete the entity successfully. Then, I try to spawn the robot the same way as in [this file](https://github.com/AcutronicRobotics/MARA/blob/master/mara_utils_scripts/scripts/spawn_entity.py) or even using the LaunchService -> LaunchDescription containing a node that launches the same `spawn_entity` file. But the result is the same, gazebo process dies or the robot simply is not spawning.

Any possible suggestion is welcomed.

---
# Working solution for simple models.
The solution seems pretty simple, and works with simple gazebo models. Example:

```python3
# delete the model.
delete_cli = self.node.create_client(DeleteEntity, '/delete_entity')
while not delete_cli.wait_for_service(timeout_sec=1.0):
    self.node.get_logger().info('service not available, waiting again...')

delete_request = DeleteEntity.Request()
delete_request.name = "target"
future = delete_cli.call_async(delete_request)
rclpy.spin_until_future_complete(self.node, future)

# spawn the model
spawn_cli = self.node.create_client(SpawnEntity, '/spawn_entity')
while not spawn_cli.wait_for_service(timeout_sec=1.0):
    self.node.get_logger().info('service not available, waiting again...')

    model_xml = "<?xml version=\"1.0\"?> \
                    <robot name=\"myfirst\"> \
                      <link name=\"world\"> \
                      </link>\
                      <link name=\"cylinder0\">\
                        <visual>\
                          <geometry>\
                            <sphere radius=\"0.01\"/>\
                          </geometry>\
                          <origin xyz=\"0 0 0\"/>\
                          <material name=\"rojotransparente\">\
                              <ambient>0.5 0.5 1.0 0.1</ambient>\
                              <diffuse>0.5 0.5 1.0 0.1</diffuse>\
                          </material>\
                        </visual>\
                        <inertial>\
                          <mass value=\"5.0\"/>\
                          <inertia ixx=\"1.0\" ixy=\"0.0\" ixz=\"0.0\" iyy=\"1.0\" iyz=\"0.0\" izz=\"1.0\"/>\
                        </inertial>\
                      </link>\
                      <joint name=\"world_to_base\" type=\"fixed\"> \
                        <origin xyz=\"0 0 0\" rpy=\"0 0 0\"/>\
                        <parent link=\"world\"/>\
                        <child link=\"cylinder0\"/>\
                      </joint>\
                      <gazebo reference=\"cylinder0\">\
                        <material>Gazebo/GreenTransparent</material>\
                      </gazebo>\
                    </robot>"

spawn_request = SpawnEntity.Request()
spawn_request.name = "target"
spawn_request.xml = model_xml
spawn_request.robot_namespace = ""
spawn_request.initial_pose = self.temp_var
spawn_request.reference_frame = ""

future = spawn_cli.call_async(spawn_request)
```

chapulina, 2018-12-31:
The reset services used to be part of `gazebo_ros_api_plugin`, which is being split into multiple plugins as we migrate to ROS 2, see https://github.com/ros-simulation/gazebo_ros_pkgs/issues/779. If you need it urgently, you're welcome to migrate them to `gazebo_ros_init` yourself and I can give a hand once I'm back from vacation.

@nzlz, please ticket a new issue for the spawn problems you're having.

chapulina, 2019-01-02:
@LuisYordano , I can work on this later today if you haven't started yet.

LuisYordano, 2019-01-02:
Hi **@chapulina**, thanks for your answer, you can start, because I am still research the new features simulation in gazebo, ros2 and rviz2.

chapulina, 2019-08-14:
Done on #866

closed 2019-08-14 by chapulina