ros-simulation/gazebo_ros_pkgs #864: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name simple_arm

horagong, 2018-12-31:
I am using ROS melodic with python3 on macOS.

```
<launch>
  <!--Include description and control launch files -->
  <include file="$(find simple_arm)/launch/robot_description.xml"/>
  <include file="$(find simple_arm)/launch/robot_control.xml"/>
  
  <!--Launch a gazebo world -->
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find simple_arm)/worlds/willow_garage.world"/>
    <arg name="paused" value="false"/>
    <arg name="use_sim_time" value="true"/>
    <arg name="gui" value="true"/>
    <arg name="headless" value="false"/>
    <arg name="debug" value="false"/>
    <arg name="verbose" value="true"/>
  </include>

  <!--spawn a robot in gazebo world-->
  <node name="urdf_spawner" pkg="gazebo_ros" type="spawn_model" respawn="false" output="screen"
  args="-urdf -param robot_description -x 0 -y 0 -z 0 -R 0 -P 0 -Y 0 -model simple_arm"/>
</launch>
```
I got the error mesage when `roslaunch simple_arm this_launch_file`
(https://github.com/udacity/simple_arm)
```
[INFO] [1546261771.539477, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1546261771.546879, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1546261771.854520, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1546261771.934455, 261.179000]: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name simple_arm
[ERROR] [1546261771.934914, 261.179000]: Spawn service failed. Exiting.
```
But when I run two launch files seperately, it is ok.
One launch file except spawn_model part and the other launch file only with spawn_model part.
```
[INFO] [1546266731.008176, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1546266731.019786, 277.090000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1546266731.025019, 277.096000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1546266731.289585, 277.286000]: Spawn status: SpawnModel: Successfully spawned entity
[urdf_spawner-1] process has finished cleanly
```

zhengSHUAI915, 2019-06-04:
I have the same problem as you,have you fixed it?

marcin-sielski, 2019-12-17:
I have fixed this issue with following PR:
https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1024

wakoko79, 2022-08-17:
I may have the solution for this, and it is absurdly simple:
The world file you launched has sim time embedded on it (when someone saved that world file, the sim time was also saved in it).

So, **go to the world file, find "sim_time" and change its value to 0**. It doesn't matter if you also change the "real_time" since it will reset itself upon gazebo start-up.

The worst thing about this issue is that it really is not an issue. Gazebo will still simulate and everything will work, but that error message will haunt you forever even there really are no issues.

I tried delaying the spawn_model node prior to this solution for like 5 seconds [link](https://answers.ros.org/question/233353/set-delay-between-starting-nodes-within-launch-file/).
**It worked**, but another warning related to ekf_localization just won't go away, so I still can't let it go.

My ekf_localization node throws an error:
`Failed to meet update rate! Took 4555.0000000001`

It took me hours just to realize that the sim time was not reset, then I edited the world file and the warning from ekf_localization went away.
Then I thought that this may be related to the spawn_model issue of gazebo, so I went and removed the code for delaying node   execution, and there you have it.

Sorry I went on for too long as this issue drove me nuts, I've been looking for the solution on and off for like half a year now.
It is just weird that delaying the spawn_model node still fixes this (also separate consoles to launch world and run spawn_model will not have this issue).

I hope this helps.

shrinivas96, 2023-09-15:
> So, **go to the world file, find "sim_time" and change its value to 0**. It doesn't matter if you also change the "real_time" since it will reset itself upon gazebo start-up.

@wakoko79 So just to confirm, I changed the line that looked like this:
```
<state world_name='default'>
      <sim_time>8278 877000000</sim_time>
      <real_time>2388 716007937</real_time>
      <wall_time>1483584500 175380351</wall_time>
```

to this:

```
<state world_name='default'>
      <sim_time>0 0</sim_time>
      <real_time>2388 716007937</real_time>
      <wall_time>1483584500 175380351</wall_time>
```

The error is gone now, but any ideas why the spawning of a robot fails due to changes in environment world files (not sure if this is the same for everyone, but in mycase the robot spawn node dies.)

Also some times I can see the robot is present in the gazebo simulation, and i can see the laser scan for example also in rviz. Not sure what fails in that case?