ros-simulation/gazebo_ros_pkgs #865: [ros2] Can't respawn robot using /spawn_entity

nzlz, 2018-12-31:
Hello everyone,

As part of a Reinforcement Learning task using gazebo I need to be able to reset the pose of a robot, [MARA](https://github.com/AcutronicRobotics/MARA) in this case. The normal behavior of the process won't be to reset the model, but one of the use cases is that the robot gets stuck after selecting an undesired action (Trajectory). Since the robot will lose plenty of time while its stuck, it is convenient that the pose of the robot is reset to the initial one.

My first idea has been to first delete the robot entity, and then spawn it again. But I am struggling with the re-spawning part. Know that **I am completely open to any different approach**, like setting poses manually or anything. 

Quoting myself from [this](https://github.com/ros-simulation/gazebo_ros_pkgs/issues/863#issuecomment-450655312) issue:

> My problems come when I try to spawn the [MARA](https://github.com/AcutronicRobotics/MARA) back again. First, I delete the entity successfully. Then, I try to spawn the robot the same way as in [this file](https://github.com/AcutronicRobotics/MARA/blob/master/mara_utils_scripts/scripts/spawn_entity.py) or even use the LaunchService -> LaunchDescription containing a node that launches the same `spawn_entity` file. But the result is the same, gazebo **process dies or the robot simply is not spawning**.

Just to clarify, I manage to spawn MARA successfully in the initialization. But not after that. I have tried manually pausing the simulation while the re-spawning has to occur, but it has no effect.

I have to mention that (gazebo+ros) are running in a different process. The learning script communicates with them successfully anyway.

```python3
    # Create the LauchService and feed the LaunchDescription obj. to it.
    ls = LaunchService()
    ls.include_launch_description(ld)
    p = Process(target=ls.run)
    p.start()
```

**Note**: Simple models re-spawn successfully ([issue](https://github.com/ros-simulation/gazebo_ros_pkgs/issues/863#issuecomment-450655312)), but they are not initially launched in the separate process. 

### Setup
Ubuntu 18.04, Gazebo 9.6.0, ROS Crystal.

chapulina, 2019-01-03:
If you run Gazebo in verbose mode, do you see any helpful error messages before it crashes?

Also, could you try getting a backtrace using gdb? Run `gdb gzserver` instead of `gazebo`, and you can run `gzclient` on a separate terminal if you need the GUI.

nzlz, 2019-01-03:
I updated the comment above (section _Port time commands (pause / reset)_). I will ask in gazebo answers once https://github.com/ros-simulation/gazebo_ros_pkgs/pull/866 is accepted, as I don't think this is related to the new `reset_simulation` code.

chapulina, 2019-01-03:
I responded some of your points on Gazebo Answers. Let me address the other ones which are here but not there:

> I'm curious why reset_simulation manages to reset the model but not reset_world.

I think the name of these 2 functionalities are not descriptive enough and may cause a lot of confusion. Looking at the [implementation](https://bitbucket.org/osrf/gazebo/src/582827025ea59c832cc50e33d64a07ae13cad76a/gazebo/physics/World.cc?fileviewer=file-view-default#World.cc-1273) inside Gazebo, these are the effects of the 2 services:

* `reset_world`: Calls `World::ResetEntities`, which resets the pose, velocities and forces of all models, their links and joints, as well as their plugins.
* `reset_simulation`: Calls `World::Reset`, which besides calling `ResetEntities`, also resets:
    * The random seed
    * Simulation time and real time
    * Calls `Reset` on all world plugins
    * Fires the `worldReset` event

So the reason why one affects your model but not the other comes from how your plugin that sets forces is handling these all these effects.

> gdb gzserver! I can't launch gazebo this way

It looks like you're using `roslaunch`? Try just `gdb gzserver` in a terminal, and inside `gdb`, do `r --verbose -s libgazebo_ros_factory.so /home/nestor/ros2_mara_ws/install/share/mara_gazebo/worlds/empty_speed_up.world`.

nzlz, 2019-01-04:
First of all thanks for your help, from your explanations and some test I have done I'm almost 100% sure we have a problem with the mara joint plugin. So I will follow this error in gazebo answers (will answer there after this).

Anyway, just in case I managed to get the log. There are 3 scenarios after a failing spawn-process:

1. No error, but no spawn neither.
2. 
```bash
malloc_consolidate(): invalid chunk size
munmap_chunk(): invalid pointer
[ERROR] [launch]: process[gazebo-1] process has died [pid 12629, exit code 255, cmd 'gazebo --verbose -s libgazebo_ros_factory.so -s libgazebo_ros_init.so /home/nestor/ros2_mara_ws/install/share/mara_gazebo/worlds/empty_speed_up.world'].
```
3. 
```bash
gzserver: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::physics::Model; typename boost::detail::sp_member_access<T>::type = gazebo::physics::Model*]: Assertion `px != 0' failed.
[ERROR] [launch]: process[gazebo-1] process has died [pid 13145, exit code 255, cmd 'gazebo --verbose -s libgazebo_ros_factory.so -s libgazebo_ros_init.so /home/nestor/ros2_mara_ws/install/share/mara_gazebo/worlds/empty_speed_up.world'].
```

I don't think these errors point to anywhere or help at all, so everything must be related to a bad failing plugin on our side. 

This issue can be closed on my side, please feel free to close!

closed 2019-01-14 by nzlz