ros-simulation/gazebo_ros_pkgs #867: [ros2] Error - Install gazebo_ros_pkgs on Ubuntu 18.04

LuisYordano, 2019-01-03:
Hello everyone,

I have followed the step by step instructions to install "gazebo_ros_pkgs" from the source code: "http://gazebosim.org/tutorials?tut=ros2_installing&cat=connect_ros" :

[image]

-----------------------------------

The error is as follows:

```
--- stderr: gazebo_msgs                               
Message interface 'geometry_msgs/Wrench' contains an unknown field type: geometry_msgs/Wrench[] wrenches
```
--------
Setup
Ubuntu 18.04, Gazebo 9.6.0, ROS Crystal.

Thanks in advance for your answer

chapulina, 2019-01-03:
There seems to be something wrong with your `geometry_msgs` package, `geometry_msgs/Wrench` [should not have](https://github.com/ros2/common_interfaces/blob/master/geometry_msgs/msg/Wrench.msg) a `wrenches` field.

How did you install it? If it was from debian packages, can you check where you got it from, like this:

~~~
$ apt-cache policy ros-crystal-geometry-msgs
ros-crystal-geometry-msgs:
  Installed: 0.6.1-0bionic.20181214.022922
  Candidate: 0.6.1-0bionic.20181214.022922
  Version table:
 *** 0.6.1-0bionic.20181214.022922 500
        500 http://repo.ros2.org/ubuntu/main bionic/main amd64 Packages
        100 /var/lib/dpkg/status
~~~

LuisYordano, 2019-01-04:
Hi @chapulina,

I have reinstalled everything again and the error persists, it is not possible to compile gazebo_ros_pkgs:
- sudo apt install ros-crystal-desktop
- sudo apt install ros-crystal-gazebo-ros-pkgs

[image]
------
```
luis@robtech:~/ROS2/ws_gazebo2$ ls src/
common_interfaces  gazebo_ros_pkgs
luis@robtech:~/ROS2/ws_gazebo2$ colcon build --symlink-install
Starting >>> std_msgs
Starting >>> std_srvs
Starting >>> gazebo_dev
Finished <<< gazebo_dev [1.91s]                                                                               
Finished <<< std_srvs [26.6s]                                                                   
[Processing: std_msgs]                              
Finished <<< std_msgs [1min 19s]                                 
Starting >>> geometry_msgs
Starting >>> actionlib_msgs
Finished <<< actionlib_msgs [16.0s]                                                                            
[Processing: geometry_msgs]                                   
[Processing: geometry_msgs]                                          
Finished <<< geometry_msgs [1min 18s]                                 
Starting >>> sensor_msgs
Starting >>> trajectory_msgs
Starting >>> nav_msgs
Starting >>> diagnostic_msgs
Starting >>> shape_msgs
Starting >>> visualization_msgs
[Processing: diagnostic_msgs, nav_msgs, sensor_msgs, shape_msgs, trajectory_msgs, visualization_msgs]                                     
Finished <<< shape_msgs [47.5s]                                                                                                             
Finished <<< trajectory_msgs [53.5s]                                                                                                            
Starting >>> gazebo_msgs
Finished <<< diagnostic_msgs [57.1s]                                                                                                            
Finished <<< visualization_msgs [1min 23s]                                                                                                      
Finished <<< nav_msgs [1min 27s]                                                                                                                
[Processing: gazebo_msgs, sensor_msgs]                                                                            
[Processing: gazebo_msgs, sensor_msgs]                                                                             
Finished <<< sensor_msgs [2min 53s]                                                                                   
Starting >>> stereo_msgs
Finished <<< stereo_msgs [14.4s]                                                                                 
Starting >>> common_interfaces
Finished <<< common_interfaces [0.69s]                                                                           
[Processing: gazebo_msgs]                                         
[Processing: gazebo_msgs]                                            
Finished <<< gazebo_msgs [3min 20s]                                  
Starting >>> gazebo_ros
[Processing: gazebo_ros]                                               
--- stderr: gazebo_ros                                       
CMake Warning at CMakeLists.txt:67 (add_library):
  Cannot generate a safe runtime search path for target gazebo_ros_factory
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libconsole_bridge.so.0.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib

  Some of these libraries may not be found correctly.

CMake Warning at CMakeLists.txt:51 (add_library):
  Cannot generate a safe runtime search path for target gazebo_ros_init
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libconsole_bridge.so.0.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib

  Some of these libraries may not be found correctly.
  ...
```
You could publish the debian package from **gazebo_ros_pkgs** with the new enhancement ros2 (reset / pause)?

chapulina, 2019-01-04:
Humm your error looks different now...

That pull request needs to be approved before it can be merged and released. If your team is using it and it works for you, feel free to add your approval.

Is it ok to close this issue, as it seems to be caused by a local issue on your setup?

LuisYordano, 2019-01-04:
Ok, @chapulina, I close this issue. I wait for the new released the **gazebo_ros_pkgs**  with the new enhancement ros2 (reset / pause).
Thank you.

closed 2019-01-04 by LuisYordano