ros-simulation/gazebo_ros_pkgs #872: Camera intrinsic parameters should be taken from plugin

christian-rauch, 2019-01-14:
The Gazabo camera plugin specifies the camera intrinsic parameters (`center`, `horizontal_fov`, `k1`, ...) for the correct projection.
The ROS publisher for the camera sensor plugins (`gazebo_ros_camera`, `gazebo_ros_depth_camera`, ...) has its own set of intrinsic parameters (`Cx`, `focalLength`, `distortionK1`, ...) which are only used for populating and publishing the CameraInfo message. A user of these ROS plugins needs to maintain these parameters in two places for no reason.

It would be easier and less error-prone, if the ROS plugins would directly reuse the camera parameters specified in the camera plugin than to specify them again in a second place.

yamokosk, 2020-04-22:
I have just realized this myself! Insanity. And then other params like `cxPrime` are simply unused. And don't get me started on the multicamera plugin... It seems to me that these camera plugins could use a good dose of TLC.

Only use case a colleague brought up is to model slightly incorrect camera params. But absolutely the default should be for the ros plugin to pull information from the gazebo sensor. If I get some time in the coming weekends, I'll submit a PR to address this, unless there is some non-obvious reason to keep it this way.

chapulina, 2020-04-22:
I believe that's what the `auto_distortion` element was created for, see https://github.com/ros-simulation/gazebo_ros_pkgs/pull/761. And on ROS 2, the parameters can only be specified on the camera as described [here](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-Camera#shared).

yamokosk, 2020-04-22:
@chapulina Where is the documentation for the ros plugin XML parameters? Other than reading the code.

No I have not seen this and it looks like it only covers distortion and not lens parameters. But again, haven't read the code to know for sure.

chapulina, 2020-04-22:
> Where is the documentation for the ros plugin XML parameters?

That should be documented [here](http://gazebosim.org/tutorials?tut=ros_gzplugins&cat=connect_ros#Camera), but it isn't. When you go over the code, it would be great if you could update the docs (but note that the `gazebo_tutorials` repository is being migrated from BitBucket to GitHub and we're not taking pull requests this week).

> it looks like it only covers distortion and not lens parameters

If that's the case, I think a similar approach of adding an `auto_*` flag could be handy here.

Also note that Gazebo itself only supports camera intrinsics since [version 9.9](https://github.com/osrf/gazebo/blob/gazebo11/Changelog.md#gazebo-990-2019-05-23).

yamokosk, 2020-04-23:
Thanks @chapulina, but that documentation is more narrative/tutorial in form. While useful in some usecases, in others a complete listing of parameters would be preferred (e.g. http://sdformat.org/spec?ver=1.7&elem=sensor#sensor_camera). I don't have a good suggestion for how to address this at the moment, but if I come up with one, I'll post on a new issue.