ros-simulation/gazebo_ros_pkgs #873: Incorrect lighting in gazebo_ros_openni_kinect

christian-rauch, 2019-01-15:
When defining a depth sensor (`<sensor type="depth" name="camera">`) and using the `libgazebo_ros_openni_kinect` plugin, I am getting images with incorrect lighting (no reflections, no shadows):
[image]

With the camera sensor (`<sensor type="camera" name="camera">`) and the `libgazebo_ros_camera` plugin I am getting corectly illuminated images:
[image]

Is this a bug that has been fixed in the original camera, but not backported to the openni plugin?

christian-rauch, 2019-01-15:
Here is a minimal world file to reproduce the issue: [camera_lighting_problem.world.txt](https://github.com/ros-simulation/gazebo_ros_pkgs/files/2760433/camera_lighting_problem.world.txt)

1. `rosrun gazebo_ros gazebo [...]/camera_lighting_problem.world` and observe that camera images are published without shadows
2. uncomment the `camera` sensor and comment the `depth` sensor
3. uncomment the `libgazebo_ros_camera` sensor and comment the `libgazebo_ros_openni_kinect` sensor
4. run again and observe that the published images have the correct lighting

christian-rauch, 2019-01-15:
It turns out that you can also get the correct lighting for the openni/depth sensor, if you just add another camera.
For example: [camera_lighting_problem.world.txt](https://github.com/ros-simulation/gazebo_ros_pkgs/files/2760704/camera_lighting_problem.world.txt). If you uncomment the camera sensor, the correct lighting is published on the depth sensor.

shiveshkhaitan, 2019-03-08:
The minimal world example does not seem to have lighting at all. Can you give another example of a world with proper lighting in which you are facing the same problem?

christian-rauch, 2019-03-08:
I am not sure what you mean. The `camera_lighting_problem.world` that I attached has two light sources defined: the `sun` as `directional` light source and `user_point_light_0` as `point` light.

shiveshkhaitan, 2019-03-09:
This is what is get in gazebo when I launch the world in either configuration.

[image]

The camera images for both of the configurations are exactly the same.

christian-rauch, 2019-03-11:
In your screenshot, there seems to be an issue with lighting. I.e. there is less illumination and it may therefore appear to give similar results. When I use the world file, I get much stronger lighting and it also propagates differencetly to the published image (depending on the sensor).

I created two distinct world configuration files with the same scene, but different sensors.

The world [lighting_problem_ros_camera.world.txt](https://github.com/ros-simulation/gazebo_ros_pkgs/files/2952006/lighting_problem_ros_camera.world.txt) uses the `libgazebo_ros_camera` plugin and publishes this image:
[image]

The world [lighting_problem_openni_kinect.world.txt](https://github.com/ros-simulation/gazebo_ros_pkgs/files/2952009/lighting_problem_openni_kinect.world.txt) uses the `libgazebo_ros_openni_kinect` plugin and publishes this image:
[image]

I am using Gazebo 10 and `gazebo_ros_pkgs` 2.5.18.

At the moment, it is hard for me to tell if this is an issue in the Gazebo plugin or the ROS package. The topic visualiser in Gazebo shows the same colour image for the `ros_camera` sensor, but it only shows the depth image for the `openni_kinect` sensor. See also the issue in the Gazebo repository: https://bitbucket.org/osrf/gazebo/issues/2584/incorrect-illumination-in-camera-sensor.

khs-sm, 2021-10-15:
Hi @christian-rauch, did you managed to solve it? The link to bitbucket seem died, will appreciate for the suggestion on this