ros-simulation/gazebo_ros_pkgs #884: [ros2] Build errors in Windows

jnoyola, 2019-02-06:
I'm not sure if there's much Windows development going on here, but the ros2 branch _almost_ builds in Windows 10, apart from a few conflicting `#define`s and missing `usleep`. I would open a PR, but I think my fixes might be a little hacky so I wanted to get some opinions.

1. `OUT` [is a macro in Windows](https://stackoverflow.com/questions/35907493/in-and-out-macros-in-minwindef-h), so I simply changed all template tparams from `OUT` to `T`.
2. ROS2 and Gazebo each build fine on their own in Windows, but we see an issue with connecting them here because Gazebo includes `<windows.h>` and ROS uses `NO_ERROR` as an enum value, which is a macro in Windows. I think it should be the consumer's responsibility to make sure they don't clash, so I `#undef NO_ERROR` before including and tf2 headers from ROS.
3. `usleep` does not exist in Windows (luckily gazebo_ros_pkgs was only relying on unistd.h indirectly, so no issue there). In gazebo_ros/utils.hpp I added a simple macro for `usleep` in Windows.
4. I also noticed that ImuSensorPlugin isn't linked to gazebo_ros_imu_sensor, but I don't think that's related to Windows.

If it would help, I can get a branch with these changes.

chapulina, 2019-02-06:
Thanks for all the pointers, a pull request would be appreciated

jnoyola, 2019-02-09:
Thanks for the quick review! Building successfully for me on Windows 10 now

closed 2019-02-09 by jnoyola