ros-simulation/gazebo_ros_pkgs #886: gazebo_ros_control [ERROR] [...]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains

matteolucchi, 2019-02-11:
Hello, I am running a simulation of a mobile manipulator, composed by a mobile base with differential drive and a 6 DOF manipulator. 

For the manipulator I am using a `velocity_controllers/JointGroupVelocityController` controller, whereas for the mobile base I am using `diff_drive_controller/DiffDriveController`. 
After the latest update fa9c99042d1f013398d4fa68398ca772283b85b8 , this error started to pop up for all the joints in my robot model 

`[ERROR] [1549875853.096599133, 0.167000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/<joint_name>`. 

The controllers still work fine, but I don't understand what causes this error. I tried adding pid gains to the yaml configuration file of the controllers:

```
/gazebo_ros_control:
  pid_gains:
      joint_name:
          p: 100.0
          i: 1.0
          d: 0.1
```
for all the joints. Adding this removes the error when launching the simulation but produces weird behavior for the robot. The robot arm is not standing correctly anymore.
I looked around a lot but I couldn't figure out a way to solve this, this appeared only after the last update. Any hint on how to solve this? 

I am running the simulation on Gazebo 7 and Kinetic.

Thank you in advance for your time

yangtfu, 2019-02-13:
same issue

francofusco, 2019-02-13:
I had the same question some weeks ago, and I think I understood what is happening. The point is that Gazebo allows you to use position and velocity control interfaces in a flexible way:
- if you do not specify the gains: when setting the position (velocity) command, the function `SetPosition` (`SetParam`) is used (see [here](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/fa9c99042d1f013398d4fa68398ca772283b85b8/gazebo_ros_control/src/default_robot_hw_sim.cpp#L322) and [here](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/fa9c99042d1f013398d4fa68398ca772283b85b8/gazebo_ros_control/src/default_robot_hw_sim.cpp#L363)). This should mean that the command is applied exactly, and therefore I think the dynamics will not be taken into account.
- you do specify gains: when you set a position (velocity) command, for each motor an effort is computed via the PID and the command is passed to `SetForce` (see [here](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/fa9c99042d1f013398d4fa68398ca772283b85b8/gazebo_ros_control/src/default_robot_hw_sim.cpp#L355) and [here](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/fa9c99042d1f013398d4fa68398ca772283b85b8/gazebo_ros_control/src/default_robot_hw_sim.cpp#L383)). Here you have a more realistic simulation that takes into account robot's dynamics.

The reason why you get a weird behavior is probably that the gains are not well tuned. I would also consider adding gravity compensation for the manipulator (there should be a plugin in Gazebo doing it). Otherwise, if you do not care about the dynamics, simply unset the PID gains and forget about the error!

matteolucchi, 2019-02-14:
@francofusco Thank you very much for you input! I had somehow figured that it was something like that but now is more clear! Thank you!

63445538, 2019-12-01:
Same issues.

aniruddhkb, 2020-06-12:
I would just like to leave this here. In case you are using PositionJointInterface (on Melodic/Gazebo9) and it doesn't work as expected, supply _negative_ PID values. For some reason the Gazebo system provides positive feedback instead of negative.