ros-simulation/gazebo_ros_pkgs #889: gazebo_ros_control: why not using ros::Time::now when updating controllers?

francofusco, 2019-02-13:
While implementing some custom controllers, I forgot to set the parameter `/use_sim_time` to `true` when starting Gazebo & other nodes. As a result some of my controllers misbehaved. The reason is that inside my code I frequently use `ros::Time::now()`, and at some point I also use timing info passed to `ControllerBase::update(ros::Time time, ros::Duration period)`. From [here](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/fa9c99042d1f013398d4fa68398ca772283b85b8/gazebo_ros_control/src/gazebo_ros_control_plugin.cpp#L231), I understood the problem: since Gazebo updates the controller manager by passing its simulation time, my time comparisons are performed using both simulation and system clock, therefore having a wrong behavior.

Is there a specific reason for passing the simulation time rather than `Time::now`? It seems more reasonable to me to use the latter, but maybe there are drawbacks that I am not considering...

Thanks in advance!

shiveshkhaitan, 2019-03-07:
If I get it correct, your question is why do we have something like simulation time which is different from the system clock. The answer to this is [here](http://wiki.ros.org/Clock). We use sim time because:

- It provides functionality of fast forwarding, slowing and pausing time. 

- Allows use of rosbag files

Also when we load heavier models in gazebo,  the real time factor decreases. Thus time moves slow in simulation than the system clock.  The ros nodes need to follow the simulation time. 
Hope that answers your question.

francofusco, 2019-03-08:
Thanks for your answer!

Well, my concern was more that when `/use_sim_time` is not set to `true` some troubles might arise in specific cases. As an example, in my controllers I susbscribe to a topic and every time I get a message I record when it is received. To do so, since I did not have direct access to Gazebo's sim time, in the callback I was using `Time::now()`. The problem was of course that this time was not in sync with the one in simulation. As a result, I was believing that using calls to `Time::now()` in the Gazebo plugin would make more sense as it would always return a time which is consistent with other nodes regardless the value of `/use_sim_time`.

However, your answer made me think, and in particular the last point convinced me that this would just cause more troubles, especially if in a controller I try to take a numerical integration or derivative. If the simulation is going slower or faster than the wall clock, using the system time would not make sense at all.

I believe that the lesson for me is: it makes more sense to use sim time, and I should simply remember to set `/use_sim_time` to `true` ;)

closed 2019-03-08 by francofusco