ros-simulation/gazebo_ros_pkgs #891: [ros2] Error service call /get_entity_state - Gazebo

LuisYordano, 2019-02-17:
Hi **gazebo_ros_pkgs ROS2 Team**,

I have compiled gazebo_ros_pkgs -b Crystal for testing.

I write command: `ros2 service call / get_entity_state gazebo_msgs/GetEntityState '{name: turtlebot3, reference_frame: world}'`

The values returned by the service, are different with the values of the simulation in Gazebo, are these results correct?

[image]

----
In ROS1 there are these service: **gazebo/model_states**

Could you tell me what the equivalent command is in ROS2?

Thanks in advance for your answer.

nzlz, 2019-02-17:
Could it be that the Turtlebot is moving and the GUI has not updated the latest real values? 

Note: you can skip the reference_frame in the request if you want to use world for pose reference.

LuisYordano, 2019-02-17:
Hi @nzlz,

The Turtlebot3 is not moving (stopped) in the gazebo simulation.

I have also tried to execute the command without the reference_frame and the result is the same:

`ros2 service call /get_entity_state gazebo_msgs/GetEntityState '{name: turtlebot3}'`

[image]

-----

chapulina, 2019-02-25:
That's strange, I'd expect the values to be the same. Could you try a few things?

1. Pause simulation to make sure the robot doesn't move between requests
1. With simulation paused, select `turtlebot3` on the left panel again, this should trigger a new request for the pose, so we're sure that the pose on the panel corresponds to the pose in the paused state.
1. Then while still paused, issue your ros2 request again, adding `__log_level:=debug` so we can see if any suspicious messages are printed.

chapulina, 2019-08-14:
@LuisYordano , are you still experiencing this issue or can we close it?

LuisYordano, 2019-10-04:
@chapulina , can close this issue.

closed 2019-10-14 by chapulina