ros-simulation/gazebo_ros_pkgs #898: Video Plugin Crashes multiple video plugin instance in the same world

talregev, 2019-02-27:
I am using gazebo 9 with ros melodic.
I was trying the video plugin, with one video it working, and with 2 video instance it crashes.

shiveshkhaitan, 2019-03-05:
This happens only when the ```<robotNamespace>``` of the two instances of the plugins are same.

This is because of this:
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/e42cc26a473733b05abc648181a6de1286105ed6/gazebo_plugins/src/gazebo_ros_video.cpp#L76-L79

Same namespace creates two instances of the above entity with the same name.

 Keeping different ```<robotNamespace>``` for the plugins can be a temporary solution until its fixed.

talregev, 2019-03-05:
That help!
you can create a random name for <robotNamespace> if there non provide.