ros-simulation/gazebo_ros_pkgs #908: gazebo_ros_diff_drive not working on bullet?

musamarcusso, 2019-03-21:
Hello everyone,

I did some tests using the [`kobuki`](https://github.com/yujinrobot/kobuki/tree/devel/kobuki_description) robot in Gazebo 7.14 with ROS Kinetic and replaced their `gazebo_ros_kobuki` by the `gazebo_ros_diff_drive`. 
I started Gazebo both with `ode` and `bullet` and gave the `kobuki` the following command in both scenarios

```
$ rostopic pub /cmd_vel geometry_msgs/Twist "linear:
  x: 0.5
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.3" 
publishing and latching message. Press ctrl-C to terminate
```

In `ode` the `kobuki` moves in circles as expected, but in `bullet` the joint velocities on the wheel joints are zero and remain unchanged.
Is there something I am missing to configure the plugin?

musamarcusso, 2019-07-31:
I think this has something to do with the `SetParam` function not being implemented for `BulletJoint` as seen [here](https://bitbucket.org/osrf/gazebo/src/9e18a31cda6328c1d8d4ec08b9d1b6cfcdb93585/gazebo/physics/bullet/BulletJoint.cc#lines-566) and for `SimbodyJoint` as seen [here](https://bitbucket.org/osrf/gazebo/src/9e18a31cda6328c1d8d4ec08b9d1b6cfcdb93585/gazebo/physics/simbody/SimbodyJoint.cc#lines-485).

srediske, 2020-06-05:
So there's no easy way to use kobuki with bullet?

In [this file](https://github.com/yujinrobot/kobuki/blob/devel/kobuki_description/urdf/kobuki_gazebo.urdf.xacro) are friction parameters "mu1" and "mu2", which should only work for ODE if i'm not mistaken. I just wanted to change this file before i saw your post, but now i think changing this file won't make a difference.