ros-simulation/gazebo_ros_pkgs #934: X Error of failed request:  GLXBadDrawable

AlexLuya, 2019-06-12:
Hello,my configuration is 
   ubuntu 18.04+ros2 dashing

and followed this link to setup ros2+gazebo integration:
`http://gazebosim.org/tutorials?tut=ros2_installing&branch=ros2`
 
and I noticed that there isn't 
` ros-dashing-geometry2`
in repository,but:
`ros-dashing-geometry-msgs`   and  ` ros-dashing-geometry-msgs-dbgsym` 
are available,so I installed both of them

when run:
`gazebo --verbose /opt/ros/dashing/share/gazebo_plugins/worlds/gazebo_ros_diff_drive_demo.world `

got error:
```
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.103
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.103
X Error of failed request:  GLXBadDrawable
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  5 (X_GLXMakeCurrent)
  Serial number of failed request:  49
  Current serial number in output stream:  49
```
any solution?

chapulina, 2019-06-12:
This is an issue with Gazebo, not with `gazebo_ros_pkgs`.

Please follow up on this issue: https://bitbucket.org/osrf/gazebo/issues/2229/gazebo-gui-crashes-sometimes

closed 2019-06-12 by chapulina

AlexLuya, 2019-06-13:
"sudo gazebo" seems work

catskul, 2020-05-22:
Bitbucket ->github:
https://github.com/osrf/gazebo/issues/2229

it2911, 2025-01-21:
> "sudo gazebo" seems work

This recommand worked to me.