ros-simulation/gazebo_ros_pkgs #935: gazebo_ros /usr/lib/x86_64-linux-gnu/libgazebo_common.so: undefined reference

corsaircpt, 2019-06-15:
On Ubuntu 18.04, with ROS Dashing built from sources , and trying to compile TurtleBot3 Packages, I get the following compile/link error;

Would appreciate a solution to this; 

Starting >>> dwb_controller
Finished <<< costmap_queue [2.08s]                                                                            
Starting >>> dwb_critics
Finished <<< dwb_controller [1.55s]                                                                     
Starting >>> dwb_plugins
Finished <<< dwb_critics [1.60s]                                                                         
Starting >>> camera_info_manager
Finished <<< camera_info_manager [1.09s]                                                                           
Starting >>> gazebo_ros
Finished <<< dwb_plugins [2.56s]                                                                    
Starting >>> nav2_amcl
Finished <<< nav2_amcl [1.93s]                                                                         
Starting >>> nav2_navfn_planner
Finished <<< nav2_navfn_planner [1.59s]                                                                         
Starting >>> nav2_rviz_plugins
--- stderr: gazebo_ros                                                                                  
/usr/lib/x86_64-linux-gnu/libgazebo_common.so: undefined reference to `ignition::fuel_tools::FuelClient::CachedModelFile(ignition::common::URI const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
/usr/lib/x86_64-linux-gnu/libgazebo_common.so: undefined reference to `ignition::fuel_tools::FuelClient::CachedModel(ignition::common::URI const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
/usr/lib/x86_64-linux-gnu/libgazebo_common.so: undefined reference to `ignition::fuel_tools::ClientConfig::SetUserAgent(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [test/test_gazebo_ros_factory] Error 1
make[1]: *** [test/CMakeFiles/test_gazebo_ros_factory.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< gazebo_ros	[ Exited with code 2 ]
Aborted  <<< nav2_rviz_plugins                                                 

Summary: 31 packages finished [1min 26s]
  1 package failed: gazebo_ros
  1 package aborted: nav2_rviz_plugins
  1 package had stderr output: gazebo_ros
  23 packages not processed

corsaircpt, 2019-06-15:
Out of desperation, I disabled the tests in CmakeLists.txt to get around this for now. Did this in Gazebo_plugins and gazebo_ros CMakeLists.txt

#if(BUILD_TESTING)
if(0)
  add_subdirectory(test)
  find_package(ament_lint_auto REQUIRED)
  ament_lint_auto_find_test_dependencies()
endif()

would appreciate a true solution though.

chapulina, 2019-06-15:
How did you install Gazebo? What version do you have? You can check like this: `dpkg -l | grep gazebo` and `dpkg -l | grep fuel`

corsaircpt, 2019-06-17:
Hi, 
Installation was done via Turtlebot3 instructions, 
curl -sSL http://get.gazebosim.org | sh

dpkg -l | gazebo

ii  gazebo9                                            9.9.0-1~bionic                               amd64        Open Source Robotics Simulator
ii  gazebo9-common                                     9.9.0-1~bionic                               all          Open Source Robotics Simulator - Shar
ed files
ii  gazebo9-plugin-base                                9.9.0-1~bionic                               amd64        Open Source Robotics Simulator - base
 plug-ins
ii  libgazebo9:amd64                                   9.9.0-1~bionic                               amd64        Open Source Robotics Simulator - shar
ed library
ii  libgazebo9-dev:amd64                               9.9.0-1~bionic                               amd64        Open Source Robotics Simulator - Deve
lopment Files
ii  ros-crystal-gazebo-dev                             3.2.0-0bionic.20190314.123446                amd64        Provides a cmake config for the defau
lt version of Gazebo for the ROS distribution.
ii  ros-crystal-gazebo-msgs                            3.2.0-0bionic.20190402.205344                amd64        Message and service data structures f
or interacting with Gazebo from ROS2.
ii  ros-crystal-gazebo-plugins                         3.2.0-0bionic.20190406.060313                amd64        Robot-independent Gazebo plugins for 
sensors, motors and dynamic reconfigurable components.
ii  ros-crystal-gazebo-ros                             3.2.0-0bionic.20190406.054701                amd64        Utilities to interface with Gazebo th
rough ROS.
ii  ros-crystal-gazebo-ros-pkgs                        3.2.0-0bionic.20190406.062536                amd64        Interface for using ROS with the Gaze
bo simulator.

dpkg -l | grep fuel

ii  libignition-fuel-tools1-1:amd64                    1.0.0+dfsg4-4                                amd64        Ignition fuel-tools classes and funct
ions - Shared library
ii  libignition-fuel-tools1-dev:amd64                  1.0.0+dfsg4-4                                amd64        Ignition fuel-tools classes and funct
ions - Development files