ros-simulation/gazebo_ros_pkgs #936: How to create custom ros2-gazebo-plugin?

Pritish9, 2019-06-19:
I am trying to move a robot arm in gazebo. Where the robot arm needs to rotate, any suggestions which plugin to use.

I could not find any so I thought of making my own ros2-gazebo-plugin.

The tutorial ( http://gazebosim.org/tutorials/?tut=ros_plugins ) shows how to make a custom gazebo-plugin with ros.
Any idea how to make a custom plugin for gazebo for ros2(Crystal or Dashing).

shiveshkhaitan, 2019-06-19:
To move the arm, I guess applying force would be sufficient. This can be achieved using `gazebo_ros_force`. 

Wiki available [here](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki) might be useful if you want to make a custom plugin.

Pritish9, 2019-06-21:
> To move the arm, I guess applying force would be sufficient. This can be achieved using `gazebo_ros_force`.

I tried using it but the force or torque is continuous and if stopped the gravity affects the state of the joint.
It seems to need a controller like the ros_control. Does ros2 have it yet? Or any alternative.

> Wiki available [here](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki) might be useful if you want to make a custom plugin. 

ros2 is missing ros2.h (comparing to ros/ros.h from ros1). I am still unclear to how to go about to make a plugin for this reason. Please let me know if there are any solution.

shiveshkhaitan, 2019-06-21:
> It seems to need a controller like the ros_control. Does ros2 have it yet? Or any alternative.

Unfortunately ros_control has not been ported yet

> ros2 is missing ros2.h (comparing to ros/ros.h from ros1). I am still unclear to how to go about to make a plugin for this reason. Please let me know if there are any solution.

There is no ros2.h in ROS2. You can refer to [gazebo_ros_template](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/ros2/gazebo_plugins/src/gazebo_ros_template.cpp) for ROS2. Comparing it with [gazebo_ros_template](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_plugins/src/gazebo_ros_template.cpp) in ROS1 should help you figure out the changes in ROS2

closed 2019-07-08 by Pritish9

Pritish9, 2019-07-08:
@shiveshkhaitan thanks for the ref., Do you have any estimated date for when ros_control will be available ros2?, And possibly joint or effort controller?

shiveshkhaitan, 2019-07-08:
I am not sure about that. You can follow the discussion [here](https://discourse.ros.org/t/ros-control-in-ros2-wishlist/7803/13)