ros-simulation/gazebo_ros_pkgs #946: alternative for libgazebo_ros_control.so in ROS1 ?

RyanJaehyunShim, 2019-07-05:
Hi, 

I would like to ask which plugin I can use in ROS2 instead of libgazebo_ros_control.so.
I didn't really find a good substitute for it in the following [link](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki)

It will be best if there is a demo sample that I can refer to.

--------------------------------------------------------------- [original code in ROS1]
<plugin name="ros_control" filename="libgazebo_ros_control.so">
      
      
</plugin>
---------------------------------------------------------------

Thank you in advance,
Ryan

shiveshkhaitan, 2019-07-05:
There is none as of now. `ros2_control` itself hasn't been completely ported yet. It is currently under [discussion](https://discourse.ros.org/t/ros-control-in-ros2-wishlist/7803/13)

RyanJaehyunShim, 2019-07-08:
@shiveshkhaitan 

Thank you for the prompt response. 

Then, is there any joint_state_subscriber-like plugin? I found a demo for the gazebo_ros_joint_state_publisher usage but I cannot find any plugins for the opposite function that I can use in my sdf file. 

Thank you very much,
Ryan

shiveshkhaitan, 2019-07-08:
I don't think there is any such plugin currently. Can you please explain what would you expect it to do.

RyanJaehyunShim, 2019-07-08:
@shiveshkhaitan 

Thank you for your reply. Please let me know if I am misunderstanding something. 

If you run the following [tutorial](http://gazebosim.org/tutorials?tut=ros2_installing&cat=connect_ros), you can give a command to the robot in the Gazebo simulator using the /demo/cmd_demo topic. 

I would like to the same process to joints of a manipulator and control joint angles using a plugin. (Or if I don't need to use a plugin, how can I do that without using a plugin?)

Best regards,
Ryan

shiveshkhaitan, 2019-07-08:
To apply force to joints, [`Apply Joint effort`](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-Body-Wrench-and-Joint-Effort) can be used but it is currently under review #941. You can still try using it

RyanJaehyunShim, 2019-07-08:
@shiveshkhaitan 

I just tested gazebo_ros_force_demo as your suggestion, but it is not what I am looking for.  Is there any way I can give position commands to the joints (not force or torque. Let me know if there is any misunderstanding in this sentence).

As gazebo_ros_joint_state_publisher_demo will run a node that publishers the current positions, it will be nice if there is a node that subscribes position commands this time.

Thank you very much,
Ryan

shiveshkhaitan, 2019-07-08:
Unfortunately, there is no such plugin as of now

chapulina, 2019-07-09:
The progress of porting `gazebo_ros_control` to ROS 2 is already tracked on https://github.com/ros-simulation/gazebo_ros_pkgs/issues/512, so I'll close this issue for now.

closed 2019-07-09 by chapulina

rjshim, 2019-07-10:
@chapulina 

It has been no updates about gazebo_ros_control since November of last year [(link)](https://github.com/ros-controls/ros2_control/issues/1)

Could I take it as there is no official substitute for [link](http://gazebosim.org/tutorials/?tut=ros_control) yet? 

Any advice will be greatly appreciated!
Ryan

chapulina, 2019-07-10:
> Could I take it as there is no official substitute for link yet?

Correct

> I would like to the same process to joints of a manipulator and control joint angles using a plugin.

Using purely Gazebo, you could use the [JointControlPlugin](https://bitbucket.org/osrf/gazebo/src/default/plugins/JointControlPlugin.cc) for position control using a PID, or a custom plugin that publishes commands using `gazebo::transport`.

Since the ETA on `ros2_control` is unknown, someone could make a ROS 2 interface to Gazebo's joint controllers as a new `gazebo_plugin`. It would probably be less featured than `ros_control` but it could cover several use cases, much like the `diff_drive` does.

shiveshkhaitan, 2019-07-15:
>Is there any way I can give position commands to the joints (not force or torque

[Joint Pose Trajectory](https://github.com/ros-simulation/gazebo_ros_pkgs/pull/955) might be helpful.

rjshim, 2019-07-16:
@shiveshkhaitan 

Thank you very much. I did below and tried the new plugin but I've got the following error (also stated below)

$ sudo apt update && sudo apt upgrade 

[Error Code]---------------------------------------------------------------------------
[Err] [Plugin.hh:180] Failed to load plugin libgazebo_ros_joint_pose_trajectory.so: libgazebo_ros_joint_pose_trajectory.so: cannot open shared object file: No such file or directory
---------------------------------------------------------------------------

Regards,
Ryan

shiveshkhaitan, 2019-07-16:
Actually it has not been merged yet. For now you can try out my fork [here](https://github.com/shiveshkhaitan/gazebo_ros_pkgs/tree/joint_trajectory) until this makes it to the release. You will have to source build `gazebo_plugins` and `gazebo_ros`.

rjshim, 2019-07-16:
@shiveshkhaitan 

I downloaded the source package from the link and tried to build it but my pc freezes when it is building the gazebo_plugins package (at approx 76%). I tried it several times. Would you check again if it works fine or let me know when it will be merged and released? 

Regards,
Ryan

shiveshkhaitan, 2019-07-16:
It builds fine for me.  Can you try building some other branch in the fork and see what what happens?

rjshim, 2019-07-17:
@shiveshkhaitan 

I tried to build the dashing branch but failed. I do not know why I can build it in my environment but I will just wait for it to be merged. 

Thank you! 
Ryan

rjshim, 2019-07-26:
Hi @shiveshkhaitan 

I tested the plugin and I think it needs two functions to be implemented.
1. Making joints move to goal positions smoothly (now it looks more like moving in a visualisation tool. 
2. Keeping goal positions of joints (as it is a physics engine)

The following link is a video recording of the gazebo screen when I used the plugin with my robot model on Gazebo. ([link](https://drive.google.com/file/d/1B_QBaQTDFACkDWK6agjxGAGltltS8f93/view))

Thank you,
Ryan

shiveshkhaitan, 2019-07-26:
> Making joints move to goal positions smoothly (now it looks more like moving in a visualisation tool.

Since this was just a port from ROS1, I did not try to modify its behaviour from what it originally did. Smooth movements would require pid on the joints which would have to be tuned differently for all joints. Maybe we can add this feature and make it an sdf parameter for the users to choose between smooth or immediate movement. @chapulina what do you think about this

> Keeping goal positions of joints (as it is a physics engine)

Sorry I could not understand what you meant by this

chapulina, 2019-07-26:
Changing that plugin to do PID control goes beyond the scope of porting it to ROS 2. If we'd like to offer a PID option, I reiterate my suggestion from https://github.com/ros-simulation/gazebo_ros_pkgs/issues/946#issuecomment-509867527 of creating a new plugin which is a wrapper around `JointControlPlugin`.

rjshim, 2019-07-29:
@chapulina 

Thank you for the advice. I do not understand what that means. I understood that comment as there is a custom plugin that I can possibly use but there is no ROS2 interface yet so I cannot use it yet.

If I can use it, is there any reference I can refer to to better understand the usage of it. I briefly googled 'ROS2 JointControlPlugin' but I couldn't find any good source I could get info from. 

Regards,
Ryan

chapulina, 2019-07-29:
>  I understood that comment as there is a custom plugin that I can possibly use

No, it doesn't exist yet. I said I was suggesting 

> creating a new plugin which is a wrapper around JointControlPlugin

If you're not interested in implementing it, you'll need to wait for `ros2_control`. Meanwhile, you can use ROS 1's `gazebo_ros_pkgs` with the ROS 1 - ROS 2 bridge.