ros-simulation/gazebo_ros_pkgs #949: linker error for bullet soft body

Karsten1987, 2019-07-07:
When compiling the `gazebo_ros_pkgs` on mojave, I run into the following linker errors:

```
[14%] Linking CXX shared library libgazebo_ros_node.dylib
[ 17%] Linking CXX shared library libgazebo_ros_utils.dylib
ld: library not found for -lBulletSoftBody
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ld: library not found for -lBulletSoftBody
make[2]: *** [libgazebo_ros_node.dylib] Error 1
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [CMakeFiles/gazebo_ros_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [libgazebo_ros_utils.dylib] Error 1
make[1]: *** [CMakeFiles/gazebo_ros_utils.dir/all] Error 2
make: *** [all] Error 2
--- stderr: gazebo_ros
ld: library not found for -lBulletSoftBody
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ld: library not found for -lBulletSoftBody
make[2]: *** [libgazebo_ros_node.dylib] Error 1
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [CMakeFiles/gazebo_ros_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [libgazebo_ros_utils.dylib] Error 1
make[1]: *** [CMakeFiles/gazebo_ros_utils.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< gazebo_ros	[ Exited with code 2 ]
```

I am a bit puzzled about it because just before, it correctly finds the bullet dependencies:
```
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 2.88
```
And indeed, it seems like bullet is correctly installed:
```
 ➭ locate libBulletSoftBody
/usr/local/Cellar/bullet/2.88/lib/libBulletSoftBody.2.88.dylib
/usr/local/Cellar/bullet/2.88/lib/libBulletSoftBody.a
/usr/local/Cellar/bullet/2.88/lib/libBulletSoftBody.dylib
/usr/local/lib/libBulletSoftBody.2.88.dylib
/usr/local/lib/libBulletSoftBody.a
/usr/local/lib/libBulletSoftBody.dylib
```

any idea on what's going wrong here?

Karsten1987, 2019-07-07:
while debugging, I added the hardcoded path the bullet and this seemed to work:
```
link_directories("/usr/local/lib/")
```
This made me wonder where this dependency comes in in the first place, which I couldn't resolve. I believe there is the dependency of `bullet` not correctly exported.

chapulina, 2019-07-09:
Gazebo should be the one pulling Bullet, [here](https://bitbucket.org/osrf/gazebo/src/9e18a31cda6328c1d8d4ec08b9d1b6cfcdb93585/cmake/SearchForStuff.cmake#lines-531).

Karsten1987, 2019-07-19:
So after a bit of digging, I believe I found what's going wrong here.
Gazebo does check for bullet, but does not export the bullet libraries as targets, but rather _simply_ adds a linker flag
```
add_definitions( -DLIBBULLET_VERSION=${BULLET_VERSION} )
```
That works as long as the library is being found on a default LD path. On my OSX machine it is unfortunately not the case. That's why I have to manually add the `gazebo_LIBRARY_DIRS` onto the search path. The PR #963 does exactly that.

I personally think that `link_libraries` should be avoided as much as possible and would rather hope that the cmake logic of exporting bullet could change at some point in the gazebo cmake files.

chapulina, 2019-07-22:
Thanks for the fix. Feel free to open PRs against `crystal` / `dashing` if this needs backporting.

> would rather hope that the cmake logic of exporting bullet could change at some point in the gazebo cmake files.

You could ticket an [issue for Gazebo](https://bitbucket.org/osrf/gazebo/issues/) or, even better, make a PR fixing this upstream :wink:

PR #963 (2019-07-22): set gazebo library dirs
fixes #949 
More discussion on the referenced issue
closed 2019-07-22 by chapulina