ros-simulation/gazebo_ros_pkgs #964: [ros2] Spawn-Entity Node Missing?

TomShkurti, 2019-07-23:
[This wiki page](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-Spawn-and-delete) claims there is a convenience node called spawn-entity in gazebo-ros which works like the old spawn-model node, but when attempting to run the example command (or otherwise actually run spawn-model) no such executable is found. Calling the underlying service does still work, but that convenience node was much more, well, convenient.

chapulina, 2019-07-23:
The port is still on a pull request, #948. You're welcome to give that a try.

TomShkurti, 2019-07-24:
Yes, that works as it should. We'll just be building gazebo_ros from that fork specifically until such time as it is merged.

chapulina, 2019-07-24:
Thanks for trying it out. I'll close this issue once that's merged.

chapulina, 2019-08-14:
#948 has been merged and will be released in the next Dashing sync.

closed 2019-08-14 by chapulina