ros-simulation/gazebo_ros_pkgs #965: laserscan signals moves in RViz along with the robot.

Yousseflah, 2019-07-24:
Environment: Ubuntu 18.04, ROS Melodic.

Please note that the problem does not appear when I use the plugin differential_drive_controller.

Hi, I am using gazebo_ros_control plugin to control my robot, but when I move my robot, the laserscan signals moves also with the robot, even if the distance between the robot and the signals changes, but this prevents me to create a good map for navigation.

Here the screen for the initial state:

[image]

And here when I move the robot, you can see the signals moves also even if the objects are fixed in gazebo.

[image]

heikkisaul, 2020-03-19:
I am experiencing a similar issue, using Turtlebot3_simulation in ROS Kinetic, on Ubuntu 16.04

Yousseflah, 2020-03-19:
Hi @heikkisaul,
I've used differential_drive_controller instead of gazebo_ros_control and the problem was resolved