ros-simulation/gazebo_ros_pkgs #970: ROS 2 tests failing to find Gazebo plugins on ROS farm CI

chapulina, 2019-08-01:
The elevator and harness tests are failing in the ROS farm CI, but not locally or on the OSRF farm. It's a runtime failure because the Gazebo plugins which they inherit from cannot be found:

`libHarnessPlugin.so: cannot open shared object file: No such file or directory`

---

Note that `GazeboRosCamera` also inherits from a plugin shipped with Gazebo, but the camera tests are not running in the ROS farm because of the lack of GPU support.

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/566fde1f2aaf7b6e0b332770634475477d456325/gazebo_plugins/include/gazebo_plugins/gazebo_ros_camera.hpp#L58

---

These Gazebo plugins are part of the [gazebo9-plugin-base](https://packages.ubuntu.com/bionic/amd64/gazebo9-plugin-base/filelist) , which should be pulled with [libgazebo9-dev](https://packages.ubuntu.com/bionic/amd64/libgazebo9-dev). I assume this is being properly installed through the `libgazebo9-dev` [rosdep key](https://github.com/ros/rosdistro/blob/9e46768ff51391a341f0745ba03b22422c2079ce/rosdep/base.yaml#L1953-L1963).

What could be happening here? Are the plugins being installed, but in a place where Gazebo can't find them? I believe they must be in `LD_LIBRARY_PATH` to be loaded correctly.

Any help is appreciated, @j-rivero @nuclearsandwich

nuclearsandwich, 2019-08-01:
I took a look at this following the dependency chain to figure out why the package _should_ be installed.

* `gazebo_plugins` (the package with the failing test) -[exec_depend]-> `gazebo_dev`
* `gazebo_dev` -[exec_depend]-> `gazebo9`
* `gazebo_dev` -[build_export_depend]-> `libgazebo9-dev`.
* `libgazebo9-dev` -[Depend:]-> `gazebo9-plugin-base`

From the build logs it looks like `gazebo9-plugin-base` is installed in the devel job images (both build and install, and build and test)

You can reproduce the ROS buildfarm environment with the command

```
generate_devel_script.py https://raw.githubusercontent.com/ros2/ros_buildfarm_config/ros2/index.yaml dashing default gazebo_ros_pkgs ubuntu bionic amd64 > run.sh
```

Does the `GAZEBO_PLUGIN_PATH` need to be set when using Gazebo from the upstream Ubuntu repositories and is there a setup script in gazebo_dev taking care of that?

Depending on your workflow, you can either remove the `--rm` option from the final container and then restart it (sending a SIGSTOP to the python3 process that will run the container's default task) and `docker-exec` into it in order to reproduce and test or use the base image that the script generates to create your own reproduction with `docker-run`.

chapulina, 2019-08-08:
Fixed by #974, thanks for all the help, @nuclearsandwich !

closed 2019-08-08 by chapulina

traversaro, 2024-06-02:
I can't confirm this in a vanilla build of ROS 2 (hence I will not open a new issue) but I guess the macOS part of this issue occurred again on modern macOS systems due to `DYLD_LIBRARY_PATH` sanification.