ros-simulation/gazebo_ros_pkgs #988: gazebo doesn't stop on sigint, needs sigkill

fjulian, 2019-08-30:
ROS melodic
Ubuntu 18.04

When launching gazebo using the command `roslaunch gazebo_ros empty_world.launch`, everything works as expected. However, when trying to stop gazebo using Ctrl-C in the same terminal, the following output occurs:

```
^C[gazebo_gui-2] killing on exit
[gazebo-1] killing on exit
[gazebo_gui-2] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Unfortunately, before escalating to SIGTERM, roslaunch waits for 10 seconds. A workaround would be to change that timeout (see for example [here](https://github.com/ethz-asl/rotors_simulator/issues/370)), however this doesn't seem like a clean solution.
Does anyone know why this is happening and how the node could be modified to respect SIGINT?

musamarcusso, 2019-08-30:
It has been an issue for a while now #751

j-rivero, 2019-09-05:
Musa is right, lets follow in #751

closed 2019-09-05 by j-rivero

PRAy-d, 2020-11-23:
Hello, I am quite new to Gazebo, and I ran into something like this ([gazebo_gui-3] escalating to SIGTERM) today. I don't know why I got this, can you guide me to a solution?