ros-simulation/gazebo_ros_pkgs #991: [ros2] gazebo doesn't pick up the "out:=/scan" from turtlebot3 world file

mkhansen-intel, 2019-09-06:
Sometime this week a change has merged somewhere in ROS2 source (using ros2.repos master file), that seems to have broken Gazebo ability to rename the turtlebot3 topics like "/scan" and "/imu" (see here: https://github.com/ROBOTIS-GIT/turtlebot3_simulations/blob/ros2/turtlebot3_gazebo/models/turtlebot3_waffle/model-1_4.sdf#L160)

Now we don't see those topics but we see default ones like "/turtlebot3_laserscan/out" instead of "/scan"

Maybe this is related to the "--ros-args" changes that merged upstream? Just a guess.

To reproduce:
1. build ros2 from source using ros2.repos file. 
2. build latest ros2 branch of gazebo_ros_pkgs
3. `git clone https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git`
4. `export GAZEBO_MODEL_PATH=~/turtlebot3_simulations/turtlebot3_gazebo/models`
5. `gazebo --verbose -s libgazebo_ros_init.so ~/turtlebot3_simulations/turtlebot3_gazebo/worlds/turtlebot3_worlds/waffle.model`
6. Wait for simulation to load
7. In another window run `ros2 topic echo /scan` (nothing will echo)
8. `ros2 topic list` will not show the /scan topic, but will show /turtlebot3_laserscan/out instead
```

crdelsey, 2019-09-06:
It looks like it is related to https://github.com/ros2/rcl/pull/491

chapulina, 2019-09-06:
I think from Eloquent we may need to use this in the SDF: `<argument>-r ~/out:=scan</argument>`. If it that works, we should update all our examples and docs.

crdelsey, 2019-09-07:
That doesn't seem to be sufficient. I'm still seeing the output on /laserscan/out instead of scan

chapulina, 2019-09-07:
Oh sorry, maybe this?

```.xml
<argument>--ros-args -r ~/out:=scan</argument>
```

I hope we don't need this :sweat: 

```.xml
<argument>--ros-args</argument>
<argument>-r</argument>
<argument>~/out:=scan</argument>
```

mkhansen-intel, 2019-09-07:
Only that last method works :(

chapulina, 2019-09-07:
Ok, thanks for testing. I can add a new `<remapping>` element which prepends `--ros-args` and `-r` under the hood for convenience.

orduno, 2019-09-10:
Since these `<argument>`s are all scoped within `<ros>`, should prepending of `--ros-args` happen to any `<argument>` and include an additional `-r` for any `<remapping>`?

```xml
          <plugin name="my_imu_plugin" filename="libgazebo_ros_imu_sensor.so">
            <ros>
              <argument>~/out:=imu</argument>
            </ros>
          </plugin>
```

chapulina, 2019-09-10:
@orduno , yes, that's the current workaround.

When I have some time I want to automatically prepend `--ros-args` for all `<argument>` tags, and handle remappings separately through a `<remapping>` element. 

Also note that upstream will soon be updated to support the old method with warnings, which should ease the transition. See https://github.com/ros2/rcl/pull/495#discussion_r322977021.

shivaang12, 2019-10-08:
I also noticed the same behavior with `imu` when  I changed `<argument>~/out:=imu</argument>` to `<argument>--ros-args --remap ~/out:=imu</argument>`.

mikaelarguedas, 2019-10-22:
See https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1011 for an attempt at implementing the `<remapping>` tag

Tested on turtlebot3 burger at https://github.com/ROBOTIS-GIT/turtlebot3_simulations/commit/dc80aa57fc63eb24a6cc31233fd6b226436e9f85

chapulina, 2019-10-27:
Fixed in #1011 for Eloquent, thanks @mikaelarguedas ! I think it's a good idea to backport the `<remapping>` tag to Dashing, update all examples and make that the recommended pattern. If anyone is up for submitting a PR that would be appreciated :blush:

closed 2019-10-27 by chapulina

chapulina, 2020-05-18:
> I think it's a good idea to backport the <remapping> tag to Dashing, update all examples and make that the recommended pattern.

At least one user is mixing dashing and eloquent and getting bad results: https://github.com/osrf/gazebo_tutorials/pull/104.

Dashing still has one year of support. We could backport the `remapping` tag to ease users' lives.