ros-simulation/gazebo_ros_pkgs #1003: Cannot source setup.bash in workspace with gazebo_ros_pkgs

crdelsey, 2019-10-08:
**Using:** ros2 master branch and gazebo_ros_pkgs ros2 branch from 7 Oct 2019

**Steps to reproduce:**

1. Setup a `test_workspace` containing the `gazebo_ros_pkgs`, `image_common` and `vision_opencv`.
2. Source ros2_ws, and build the `test_workspace`
3. Source `test_workspace/install/setup.bash`

**Result:**
You get an error saying `ament_prepend_unique_value: command not found`

**Workaround:**
source  `test_workspace//install/share/gazebo_plugins/local_setup.bash` prior to 
Source  `test_workspace/install/setup.bash`

mjcarroll, 2019-10-08:
I believe that this was resolved by https://github.com/ament/ament_package/pull/94

Perhaps your workspace is out of date?  Try pulling the lastest version of `ament_package` and see if that resolves the issue for you.

crdelsey, 2019-10-08:
I do have that update to `ament_package`, but I still see this problem.

crdelsey, 2019-10-08:
I don't see the problem when I run in docker using the osrf/ros2:nightly image, so I'll do some more experiments to see what is different.

dirk-thomas, 2019-10-08:
The `ament_package` change is only relevant when building Debian packages. So from Debian this should work already. For a from-source build the shell function `ament_prepend_unique_value` is intentionally not available as of Eloquent. Simply because it was the main reason the sourcing of the environment was so slow. See #1004 for a fix which adds a descriptive version of the environment change which will work from source and also be faster.

crdelsey, 2019-10-08:
>  I don't see the problem when I run in docker using the osrf/ros2:nightly image, so I'll do some more experiments to see what is different.

It turns out I was getting the error in the docker test as well. It's just my build was ignoring the error and continued on, so I missed the message in all the output.

chapulina, 2019-10-14:
@crdelsey , please close if the issue solved for you. Thanks!

crdelsey, 2019-10-14:
It seems to be fixed. Thanks!

PR #1004 (2019-10-09): generate a .dsv file for the environment hook
Follow up of #974. Fixes #1003.

This also needs to be applied to the `eloquent` branch and released there.
closed 2019-10-14 by crdelsey