ros-simulation/gazebo_ros_pkgs #1007: Bug in gazebo_ros_diff_drive.cpp

trishantroy, 2019-10-12:
Hello,
In the differential drive code (https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_plugins/src/gazebo_ros_diff_drive.cpp), the following lines should be added after line 293:
`wheel_speed_instr_[LEFT] = current_speed[LEFT];
wheel_speed_instr_[RIGHT] = current_speed[RIGHT];`

Otherwise, an old instruction (wheel_speed_instr_) will be used by the algorithm. We should update the wheel_speed_instr_ in each update of GAZEBO.

chapulina, 2019-10-14:
Hi @trishantroy , could you elaborate on why you think the following lines are not enough?

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/daebf9c155aaadccffe4e34735566c526d39c86e/gazebo_plugins/src/gazebo_ros_diff_drive.cpp#L294-L302

trishantroy, 2019-10-16:
Hey @chapulina ,

The lines (L294-302) ensure that the wheel_speed_instr_ is within the acceleration constraints from the previous wheel_speed_instr. The correct implementation would be that the wheel_speed_instr_ is within the acceleration constraints from the current_speed.

chapulina, 2019-10-17:
I think I may be missing something, may need to look closer into the code. How is your suggestion taking acceleration into consideration?

trishantroy, 2019-10-17:
Sure. Do check the code. 

In order to satisfy the acceleration constraints, the instructed velocity (wheel_speed_instr_) should be within (current wheel speed - acceleration * update_time) and (current wheel speed + acceleration * update time).
Cases: 
i) abs(wheel_speed_ - current_speed) > acceleration * update_time:
In this case, the wheel_speed_instr_ should be equal to current_speed + sign(wheel_speed_ - current_speed) * acceleration * update_time.
ii) abs(wheel_speed_ - current_speed) <= acceleration * update_time:
In this case, the wheel_speed_instr_ should be equal to wheel_speed_ i.e. (current_speed + (wheel_speed - current_speed)

abs() and sign() represent the absolute and signum functions respectively.

chapulina, 2019-10-17:
The line you suggested adding shouldn't change anything, since `wheel_speed_instr_[LEFT]` and `wheel_speed_instr_[RIGHT]` will always receive new values on lines 294~302 anyway. Right?

trishantroy, 2019-10-17:
Those lines mean wheel_speed_instr = wheel_speed_instr + (fmin_part), right?
It should be wheel_speed_instr = current_speed_ + (fmin_part).

chapulina, 2019-10-17:
Argh you're right, sorry for the noise :innocent: Your suggestion makes sense :+1: