ros-simulation/gazebo_ros_pkgs #1015: gazebo(-ros) broken

MatthijsBurgh, 2019-11-01:
Ubuntu 16.04
ROS Kinetic
Gazebo version 7.16.0

```
amigo@amigo-P50 $ roslaunch gazebo_ros empty_world.launch debug:=true
... logging to /home/amigo/.ros/log/46d5ad52-fce7-11e9-a1cd-e4a7a05366c8/roslaunch-amigo-P50-3657.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://amigo-P50.local:33003/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/debug)
    gazebo_gui (gazebo_ros/gzclient)

ROS_MASTER_URI=http://localhost:11311

process[gazebo-1]: started with pid [3691]
process[gazebo_gui-2]: started with pid [3696]
[ INFO][/gazebo_gui][1572641818.865111694]: Finished loading Gazebo ROS API Plugin.
[ INFO][/gazebo_gui][1572641818.866478639]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from gzserver...(no debugging symbols found)...done.
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[New Thread 0x7fffce4d1700 (LWP 3805)]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[New Thread 0x7fffc97b2700 (LWP 3811)]
[New Thread 0x7fffc8fb1700 (LWP 3812)]
[New Thread 0x7fffc87b0700 (LWP 3815)]
[New Thread 0x7fffc7faf700 (LWP 3816)]
[New Thread 0x7fffc77ae700 (LWP 3817)]
[New Thread 0x7fffc6fad700 (LWP 3829)]
[New Thread 0x7fffc67ac700 (LWP 3830)]
[New Thread 0x7fffc5fab700 (LWP 3831)]
[New Thread 0x7fffc57aa700 (LWP 3832)]
[New Thread 0x7fffc4fa9700 (LWP 3833)]
[New Thread 0x7fffaffff700 (LWP 3834)]
[New Thread 0x7fffaf7fe700 (LWP 3835)]
[New Thread 0x7fffaeffd700 (LWP 3836)]
[New Thread 0x7fffae7fc700 (LWP 3837)]
[New Thread 0x7fffadffb700 (LWP 3838)]
[New Thread 0x7fffad7fa700 (LWP 3839)]
[ INFO][/gazebo][1572641820.638037188]: Finished loading Gazebo ROS API Plugin.
[ INFO][/gazebo][1572641820.639398954]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[New Thread 0x7fffacff9700 (LWP 3842)]
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  155 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  18
  Current serial number in output stream:  19
[Thread 0x7fffacff9700 (LWP 3842) exited]
[Thread 0x7fffc87b0700 (LWP 3815) exited]
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  155 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  18
  Current serial number in output stream:  19
[Thread 0x7fffc7faf700 (LWP 3816) exited]
[Thread 0x7fffae7fc700 (LWP 3837) exited]
[Thread 0x7fffaf7fe700 (LWP 3835) exited]
[Thread 0x7fffaffff700 (LWP 3834) exited]
[Thread 0x7fffc57aa700 (LWP 3832) exited]
[Thread 0x7fffc67ac700 (LWP 3830) exited]
[Thread 0x7fffc6fad700 (LWP 3829) exited]
[Thread 0x7fffc5fab700 (LWP 3831) exited]
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
terminate called recursively
[Thread 0x7fffadffb700 (LWP 3838) exited]

Thread 18 "gzserver" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffad7fa700 (LWP 3839)]
0x00007ffff6af3428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
Aborted (core dumped)
[gazebo_gui-2] process has died [pid 3696, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/amigo/.ros/log/46d5ad52-fce7-11e9-a1cd-e4a7a05366c8/gazebo_gui-2.log].
log file: /home/amigo/.ros/log/46d5ad52-fce7-11e9-a1cd-e4a7a05366c8/gazebo_gui-2*.log

(gdb) where
#0  0x00007ffff6af3428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff6af502a in __GI_abort () at abort.c:89
#2  0x00007ffff712d84d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff712b6b6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff712b701 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff712b919 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff3d28ff9 in void boost::throw_exception<boost::lock_error>(boost::lock_error const&) () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#7  0x00007fffca4ec5ea in ros::NodeHandle::construct(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) () from /opt/ros/kinetic/lib/libroscpp.so
#8  0x00007fffca4ec899 in ros::NodeHandle::NodeHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&) () from /opt/ros/kinetic/lib/libroscpp.so
#9  0x00007fffca82552c in gazebo::GazeboRosApiPlugin::physicsReconfigureThread() () from //opt/ros/kinetic/lib/libgazebo_ros_api_plugin.so
#10 0x00007ffff3d205d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#11 0x00007ffff65bb6ba in start_thread (arg=0x7fffad7fa700) at pthread_create.c:333
#12 0x00007ffff6bc541d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

```

j-rivero, 2019-11-02:
Hello Matthijs:

If the gazebo standalone execution does not work, the gazebo bitbucket issue tracker is the right place to report issues. Let's follow there.

closed 2019-11-02 by j-rivero

zp2546265641, 2024-06-24:
and so , how to solve it?