ros-simulation/gazebo_ros_pkgs #1016: spawn_entity_demo.launch.py does not work in eloquent

MartinCornelis2, 2019-11-11:
[spawn_entity.py-3] spawn_entity.py: error: unrecognized arguments: --ros-args
[ERROR] [spawn_entity.py-3]: process has died [pid 7068, exit code 2, cmd '/opt/ros/eloquent/lib/gazebo_ros/spawn_entity.py -entity demo -database double_pendulum_with_base --ros-args'].

The exact same launchfile does work in dashing could this be related to:
https://index.ros.org/doc/ros2/Tutorials/Node-arguments/ 

> ll ROS nodes take a set of arguments that allow various properties to be reconfigured. Examples include configuring the name/namespace of the node, topic/service names used, and parameters on the node. All ros specific arguments have to be specified after a --ros-args flag:

> `ros2 run my_package node_executable --ros-args ...` 

> For more details, see this design doc.
> 
> Note
> 
> Before Eloquent, --ros-args wasn’t needed.

?
If this is the case then either ros2launch requires an update and you can close this issue. Or this launchfile needs to be updated.

j-rivero, 2019-11-11:
Hello Martin:

I can confirm that the launch file is broken on Eloquent. I think that we need to fix the `spawn_entity_demo.launch.py`. Do you want to proceed with fixing and sending a PR to solve the problem? Thanks!

MartinCornelis2, 2019-11-12:
For my application I just settled on Dashing for now, I honestly don't know what to change in the launchfile to get it to work again. I tried passing the arguments in a few different ways (some quick simple guesses), but none of them worked. I don't have the time right now to dive into what changed between dashing and eloquent w.r.t. ros2 launch to fix this. Maybe someone else can pick this up?

chapulina, 2019-11-12:
I just built the `ros2` branch of this repository on top of the latest `ros-eloquent-*` debs and the launch file worked for me:

`ros2 launch gazebo_ros spawn_entity_demo.launch.py`

Maybe the problem has fixed itself recently? If you can still reproduce it, could you check if:
* `gazebo.launch.py` works?
* `gzserver.launch.py` works?
* Can you get a backtrace passing `gdb:=true`?

chapulina, 2019-11-13:
Ok, I just realized there's an `eloquent` branch and that's what's been released. I confirmed that the spawn example was not working there. I just synced the `eloquent` branch with `ros2` in d72c3205eabfaca6816b73db703b86c1b8ef7aeb and it works now. Feel free to close once you confirm.

MartinCornelis2, 2019-11-14:
I looked at the changes and I think that indeed 
```
    args_without_ros = rclpy.utilities.remove_ros_args(args)
    spawn_entity_node = SpawnEntityNode(args_without_ros)
```
These lines could solve the problem. I'll test it today and if it works on my end I'll close this issue. Thank you very much for the quick response!

MartinCornelis2, 2019-11-14:
The spawning demo works if I build the eloquent branch from source.
When can we expect these changes to be in the binary version?
I have no experience with making packages available through 
`apt install ros-[version]-[package]`
so I'm curious what kind of timeframes are normal for these kind of things :).
Again thanks for the help and I will now close the issue.

closed 2019-11-14 by MartinCornelis2

chapulina, 2019-11-14:
> When can we expect these changes to be in the binary version?

This has been merged: 
https://github.com/ros/rosdistro/pull/22985, so we just need to wait for the next eloquent sync.