ros-simulation/gazebo_ros_pkgs #1020: [ROS 2] gzserver and gzclient racing when using Gazebo ROS launch

hidmic, 2019-12-06:
## Description

When launching both `gzserver` and `gzclient` using [`gazebo.launch.py`](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/ros2/gazebo_ros/launch/gazebo.launch.py) with a large/heavy world (e.g. I personally tried with 20+ standing actors), `gzclient` GUI either hangs in a black screen or shows up seemingly out of sync w.r.t. to `gzserver` state. This does not happen if both are launched separately, `gzserver` first, with some time in between, so I'd presume it's a race.

## Steps to reproduce

TBD (soon)

chapulina, 2019-12-06:
Maybe it could be [this issue](https://bitbucket.org/osrf/gazebo/issues/2053/topic-name-mismatch-on-heavy-worlds-with). The fix to that is calling your world `default`, in case it's not that yet.

hidmic, 2019-12-09:
I found that issue before, and I verified the world was called `default`. So it'd seem it's something else. I haven't had time to put together a small repro yet though.

shiveshkhaitan, 2019-12-23:
@hidmic  I have tried running gazebo.launch.py with very heavy worlds and it worked. Would appreciate if you can provide the world you are trying to launch.

hidmic, 2019-12-23:
Ok, false alarm. It just takes a _long_ time to come up and I wasn't waiting enough. The impact seems somewhat more severe in Gazebo9.0 than in Gazebo9.11, but it eventually comes up. Also, I can only reproduce these delays when dynamically spawning actors from within a world plugin -- my simple world based repros don't suffer from this.

I'll close this one under the shared assumption that actors are simply expensive.

closed 2019-12-23 by hidmic