ros-simulation/gazebo_ros_pkgs #1030: question about urdf_spawner

fightingjack, 2020-01-09:
Hello, everyone. I encountered some problems in the process of simulating the robot arm, so I had to ask for your help.When I wrote a program named ur4_gazebo_world.launch, an error was reported in the process of roslaunch, please see the attached code for the content.I want to know if something is wrong with my node call?I tried to install gazebo_ros_pkgs, but it didn't work.
`[urdf_spawner-4] process has died [pid 11681, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -urdf -model ur4 -param ur4_description __name:=urdf_spawner __log:=/home/terry/.ros/log/1d046e54-32b2-11ea-a0f8-74e5434781a6/urdf_spawner-4.log].
log file: /home/terry/.ros/log/1d046e54-32b2-11ea-a0f8-74e5434781a6/urdf_spawner-4*.log
`
and here is part of my ur4_gazebo_world.launch code
`
  <node name="urdf_spawner" pkg="gazebo_ros" type="spawn_model" respawn="false" output="screen"
	args="-urdf -model ur4 -param ur4_description"/> `

chapulina, 2020-01-09:
Hi @fightingjack , please ask questions at http://answers.gazebosim.org/ . This issue tracker is for bug reports and feature requests. See you on Gazebo Answers!

closed 2020-01-09 by chapulina

fightingjack, 2020-01-10:
sad...