ros-simulation/gazebo_ros_pkgs #1043: Help to contribute a gazebo_ros_magnetometer plugin

Darkproduct, 2020-02-10:
To get a full simulation of our robot, we need a magnetometer sensor that publishes a `sensor_msgs/MagneticField` message. Unfortunately, there is only one magnetometer plugin out there. `gazebo_ros_magnetic` from [hector_gazebo_plugins](http://wiki.ros.org/hector_gazebo_plugins)). 
The three main reasons why I don't want to use it are:

- It outputs a `geometry_msgs::Vector3Stamped` instead of `sensor_msgs/MagneticField`.
- It doesn't use gazebo world `<magnetic_field>` tag in sdf file.
- It is implemented as `ModelPlugin` and not as `SensorPlugin`.

Gazebo has a `MagnetometerSensor` since 2015 and no ROS plugin. See [this commit](https://bitbucket.org/osrf/gazebo/pull-requests/1788/add-magnetometer-sensor-redo-of-andrews/diff) and [this discussion](https://bitbucket.org/osrf/gazebo/issues/1294/inclusion-of-magnetic-field-strength).

I used `gazebo_ros_imu` as template and created [gazebo_ros_magnetometer](https://github.com/Darkproduct/gazebo_ros_magnetometer).

urdf entry:
```
<gazebo reference="magnetometer_link">
  <sensor name="magnetometer_sensor" type="magnetometer">
    <always_on>true</always_on>
    <update_rate>100</update_rate>
    <visualize>true</visualize>
    <topic>magnetometer</topic>
    <plugin filename="libgazebo_ros_magnetometer.so" name="magnetometer_plugin">
      <topicName>/robot/imu/mag</topicName>
      <frameName>imu_link</frameName>
      <updateRateHZ>100.0</updateRateHZ>
    </plugin>
    <pose>0 0 0 0 0 0</pose>
  </sensor>
</gazebo>
```

But for some reason, my plugin never gets executed. The ros node is never initialized and therefore no ros topic gets advertised. My `GazeboRosMagneticSensor` class doesn't even get to its constructor. 

The only success I had was, with the magnetometer_sensor gazebo plugin. In Gazebo a magnetometer topic is advertised and `gz topic -e` outputs:
```
time {
  sec: 23
  nsec: 932000000
}
field_tesla {
  x: 4.8184694239418247e-06
  y: 2.3297836655578027e-05
  z: -4.1988012093698152e-05
}
```

This project needs more time than I anticipated and I'm by no means a gazebo expert and I hope some of you guys know, what I am doing wrong. I want to finally finish this and contribute it to this plugin collection.

chapulina, 2020-02-10:
That sounds very useful, thanks for working on it!

> But for some reason, my plugin never gets executed

It's hard to find out why just looking at the plugin code. The problem is probably related to how the plugin is being launched. If you describe all the steps you followed to install and use your plugin, someone else could try reproducing the problem.

Darkproduct, 2020-02-10:
Ok, here are the steps to run it:

1. Create a new catkin `ws` and initialize it
2. Clone [this repo](https://github.com/Darkproduct/gazebo_ros_magnetometer) into your `ws/src` folder
3. Build your `ws` (I use catkin_tools). If you want more information on how it is built look at the [CMakeLists.txt](https://github.com/Darkproduct/gazebo_ros_magnetometer/blob/master/CMakeLists.txt)
4. There should be a `libgazebo_ros_magnetometer.so` in `ws/devel/lib` folder

I added an `urdf` and `launch` folder to the repo for easy testing:

5. Source the `setup.bash` file in `ws/devel`
6. Run `roslaunch gazebo_ros_magnetometer gazebo.launch`. If you are on a vm, consider using `vm:=true` flag

Now, everything should work, but it doesn't. There is a gazebo topic: `/gazebo/default/robot/magnetometer_link/magnetometer_sensor/magnetometer` but no ros topic.

chapulina, 2020-02-10:
Thanks for the instructions.

I recommend you always run Gazebo on verbose mode so you can see errors. You can do that for example by passing `<arg name="verbose" value="true" />` to `empty_world.launch`.

When you do that, you can see that Gazebo is failing to load your plugin with this error:

```
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_ros_magnetometer.so: /tmp/mag_test/install/gazebo_ros_magnetometer/lib/libgazebo_ros_magnetometer.so: undefined symbol: _ZN6gazebo23GazeboRosMagneticSensor5ResetEv
```

That happens because you define a `Reset` function on the header which is not implemented on the source code.

I hope this helps. Will close this issue for now, and if you have more issues about writing your own plugin, please ask at http://answers.gazebosim.org/ . Looking forward to a PR with the contribution :smile:

closed 2020-02-10 by chapulina