ros-simulation/gazebo_ros_pkgs #1050: Cannot import name 'GazeboRosPaths' when using a custom overlay

ejodconi, 2020-02-22:
Currently I'm working on a project with a use of ROS2 Dashing and Gazebo.

I'm trying to setup my gazebo models without setting the GAZEBO_MODEL_PATH in terminal or `.bashrc`. I followed the [path migration guide](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-Gazebo-ROS-Paths) from this repo's wiki. I got an import error when my overlay is sourced and not when only dashing is sourced.

**Error**
```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): cannot import name 'GazeboRosPaths'
```

**To Reproduce**

Steps to reproduce the behavior:
1. `source /opt/ros/dashing/setup.bash`
2. `ros2 launch gazebo_ros gazebo.launch.py` <-- this works without errors
3. `source ~/ros_ws/install/local_setup.bash`
4. `ros2 launch gazebo_ros gazebo.launch.py` <-- creates the below error
```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): cannot import name 'GazeboRosPaths'
```

**Expected behavior**

Same as step 2.

**Environment (please complete the following information):**

 - OS: Ubuntu 18.04.4 LTS
 - Gazebo version: 9
 - ROS version: ROS2 Dashing
  - gazebo_ros_pkgs: apt install

shiveshkhaitan, 2020-02-23:
@ejodconi I have followed the steps in sequence in the same environment as yours but was unable to reproduce the error. It works fine for me with dashing and local setup. Can you please share your launch file and CMakeList which might be helpful in tracking the error.

ejodconi, 2020-02-23:
I have the same problem with the turtlebot3 repository. I have both the [basic](https://github.com/ROBOTIS-GIT/turtlebot3) repository as the [simulations](https://github.com/ROBOTIS-GIT/turtlebot3_simulations) repository inside my src folder. I'm unable to share my current project but the setup and structure is the same.

eliasdc, 2020-02-26:
I (same guy as the previous comment author, deleted duplicate account) was able to use the environment hooks from [paths migration guide](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-Gazebo-ROS-Paths) to accomplish the same behavior. But I would prefer the other way.

eliasdc, 2020-06-29:
I solved this issue. 
I had a scripts folder which I installed for python using `ament_python_install_package(scripts/)` which conflicted with the `from scripts import GazeboRosPaths` in the launch file.

Maybe it is a good idea to not use a common name like `scripts` as the install directory.

eborghi10, 2020-08-07:
I was able to reproduce this issue with Foxy but in my case I have all my `scripts` directories renamed and it's still not working.

This example (that contains a `script` directory only) works well:

```
mkdir -p /ws/src
cd /ws/src
git clone -b dashing-devel https://github.com/ros/xacro.git
cd /ws
colcon build
source install/setup.bash
```

So I suspect the issue arises when you create a ROS 2 package in [this way](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-Gazebo-ROS-Paths). I tried both methods and none of them worked.

In order to fix it, I copied my own `GazeboRosPaths` script and added this:

```Python
media += ":/usr/share/gazebo-11"
model += ":/usr/share/gazebo-11/models"
```

eliasdc, 2020-08-10:
> I was able to reproduce this issue with Foxy but in my case I have all my `scripts` directories renamed and it's still not working.
>
> This example (that contains a `script` directory only) works [...]

don't forget to clear your build and install directories to retest it.

eborghi10, 2020-08-10:
> don't forget to clear your build and install directories to retest it.

I'm using Docker without a persistent workspace. And the error appears after building the workspace for the first time.

808brick, 2021-03-30:
I can confirm the same issue with ROS Foxy on Ubuntu 20. I am able to launch gazebo in a separate terminal just fine, but unable to include in my packages launch file.

My package does not have a scripts folder, and is ament_cmake. The exact error is as follows:
`ImportError: cannot import name 'GazeboRosPaths' from 'scripts'`

In my launch file I tried two ways to launch gazebo, both ways resulting in the same error:

```
ExecuteProcess(
            cmd=["ros2", "launch", "gazebo_ros", "gazebo.launch.py"]
        ),
```
and 
```
IncludeLaunchDescription(
            PythonLaunchDescriptionSource([get_package_share_directory('gazebo_ros'),
            "/launch/gazebo.launch.py"]),
        ),
```

I have also attempted to delete the build/ install/ and log/ folders and rebuild from scratch without any luck.

808brick, 2021-03-30:
Alright, after some tinkering, I found a way to launch gazebo with my own launch file within my package. I use the following tag in my launch description:

`ExecuteProcess(
            cmd=['gazebo', '--verbose', '-s', 'libgazebo_ros_factory.so'],
            output='screen'),`

For those interested in seeing my full launch file that I used to launch a robot model within gazebo, I do the following:

```python
import os
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, ExecuteProcess
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node
from launch.launch_description_sources import PythonLaunchDescriptionSource

def generate_launch_description():

  use_sim_time = LaunchConfiguration('use_sim_time', default='false')
  urdf_file_name = 'urdf/camera_bot.xacro'

  print("urdf_file_name : {}".format(urdf_file_name))

  urdf = os.path.join(
      get_package_share_directory('ros2_sim_pkg'),
      urdf_file_name)

  return LaunchDescription([

        DeclareLaunchArgument(
            'use_sim_time',
            default_value='false',
            description='Use simulation (Gazebo) clock if true'),

        ExecuteProcess(
            cmd=['gazebo', '--verbose', '-s', 'libgazebo_ros_factory.so'],
            output='screen'),

        Node(
            package='robot_state_publisher',
            executable='robot_state_publisher',
            name='robot_state_publisher',
            output='screen',
            parameters=[{'use_sim_time': use_sim_time}],
            arguments=[urdf]),

        Node(
            package='joint_state_publisher',
            executable='joint_state_publisher',
            name='joint_state_publisher',
            output='screen',
            parameters=[{'use_sim_time': use_sim_time}]
            ),

        Node(
            package='gazebo_ros',
            executable='spawn_entity.py',
            name='urdf_spawner',
            output='screen',
            arguments=["-topic", "/robot_description", "-entity", "cam_bot"])
  ])

```

Only issue I have run into so far is sometimes gazebo takes a long time to start, and the spawn service times out. But I'm sure this could be easily adjusted to change the default time out time for the spawn service. 

Hope this helps someone down the line.

oenzan, 2023-01-31:
I believe it comes from PYTHONPATH  enviroment variable. I built turtlebot3 from source code in my workspace and when i try to run `ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py` command i get following errors:

`[INFO] [launch]: All log files can be found below /home/ozan/.ros/log/2023-01-31-15-45-23-253632-ozan-OptiPlex-7060-686150
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=ImportError("cannot import name 'GazeboRosPaths' from 'scripts' (/home/ozan/ros2_ws/install/rvc_application/lib/python3.8/site-packages/scripts/__init__.py)")>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 130, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_description_source.py", line 51, in _get_launch_description
    return get_launch_description_from_python_launch_file(location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 62, in get_launch_description_from_python_launch_file
    launch_file_module = load_python_launch_file_as_module(python_launch_file_path)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 37, in load_python_launch_file_as_module
    loader.exec_module(mod)
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/opt/ros/foxy/share/gazebo_ros/launch/gzserver.launch.py", line 28, in <module>
    from scripts import GazeboRosPaths
ImportError: cannot import name 'GazeboRosPaths' from 'scripts' (/home/ozan/ros2_ws/install/rvc_application/lib/python3.8/site-packages/scripts/__init__.py)
`

My PYTHONPATH env. shown below:

```
ozan@ozan-OptiPlex-7060:~/ros2_ws$ echo $PYTHONPATH
```
`
/home/ozan/ros2_ws/install/xacro/lib/python3.8/site-packages:/home/ozan/ros2_ws/install/turtlebot3_teleop/lib/python3.8/site-packages:/home/ozan/ros2_ws/install/turtlebot3_example/lib/python3.8/site-packages:/home/ozan/ros2_ws/install/turtlebot3_msgs/lib/python3.8/site-packages:/home/ozan/ros2_ws/install/rvc_coveragepp/lib/python3.8/site-packages:/home/ozan/ros2_ws/install/rvc_msgs/lib/python3.8/site-packages:/home/ozan/ros2_ws/install/rvc_motion_planning/lib/python3.8/site-packages:/home/ozan/ros2_ws/install/rvc_application/lib/python3.8/site-packages:/home/ozan/ros2_ws/install/dynamixel_sdk_custom_interfaces/lib/python3.8/site-packages:/opt/ros/foxy/lib/python3.8/site-packages
`
If i move `/opt/ros/foxy/lib/python3.8/site-packages` to head of PYTHONPATH, error disappears and It worked as it should. Some how parser use foremost path that appear in PYTHONPATH.

I also try @ghost situation.
`source ~/ros2_ws/install/setup.bash` probably cause that problem . It appends `/opt/ros/foxy/lib/python3.8/site-packages` to PYTHONPATH.

In `~/ros2_ws/install/setup.bash`
```
 20 # source chained prefixes
 21 # setting COLCON_CURRENT_PREFIX avoids determining the prefix in the sourced script
 22 COLCON_CURRENT_PREFIX="/opt/ros/foxy"
 23 _colcon_prefix_chain_bash_source_script "$COLCON_CURRENT_PREFIX/local_setup.bash"
 24 
 25 # source this prefix
 26 # setting COLCON_CURRENT_PREFIX avoids determining the prefix in the sourced script
 27 COLCON_CURRENT_PREFIX="$(builtin cd "`dirname "${BASH_SOURCE[0]}"`" > /dev/null && pwd)"
 28 _colcon_prefix_chain_bash_source_script "$COLCON_CURRENT_PREFIX/local_setup.bash"

```