ros-simulation/gazebo_ros_pkgs #1070: Change opencv header

emmanuelkring, 2020-03-29:
**Description**
I am trying to compile on Ubuntu 20.04 with noetic and I get the following error: 
```
gazebo_ros_pkgs/gazebo_plugins/src/gazebo_ros_prosilica.cpp:47:10: 
fatal error: opencv/cv.h: No such file or directory
```
I think this is because opencv 4 does not have the C header file "opencv/cv.h".
Changing it to opencv2/opencv.hpp solved the problem.

chapulina, 2020-03-30:
Can you give #1068 a try?

emmanuelkring, 2020-03-30:
It solves this issue but there is another unrelated compile error on Ubuntu 20.04 so it does not compile.
Not sure if I should open a new issue for this but the error is

>In file included from gazebo_ros_pkgs/gazebo_plugins/src/gazebo_ros_depth_camera.cpp:28:
gazebo_ros_pkgs/gazebo_plugins/include/gazebo_plugins/gazebo_ros_depth_camera.h:94:30: error: ‘virtual void gazebo::GazeboRosDepthCamera::OnNewReflectanceFrame(const float*, unsigned int, unsigned int, unsigned int, const string&)’ marked ‘override’, but does not override
   94 |      protected: virtual void OnNewReflectanceFrame(const float * _reflectance,
      |                              ^~~~~~~~~~~~~~~~~~~~~
gazebo_ros_pkgs/gazebo_plugins/include/gazebo_plugins/gazebo_ros_depth_camera.h:99:29: error: ‘virtual void gazebo::GazeboRosDepthCamera::OnNewNormalsFrame(const float*, unsigned int, unsigned int, unsigned int, const string&)’ marked ‘override’, but does not override
   99 |     protected: virtual void OnNewNormalsFrame(const float * _normals,
      |                             ^~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/gazebo_ros_depth_camera.dir/build.make:63: CMakeFiles/gazebo_ros_depth_camera.dir/src/gazebo_ros_depth_camera.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:4544: CMakeFiles/gazebo_ros_depth_camera.dir/all] Error 2

chapulina, 2020-03-30:
For that one you'll need the latest Gazebo 9 release. Some `ifdef`s could be added to the code so people can build the noetic branch against Gazebo <= 9.12

FYI @ahcorde

ahcorde, 2020-03-31:
Added the preprocessor conditions https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1071

emmanuelkring, 2020-03-31:
Great it works now. Thank you!

closed 2020-03-31 by chapulina