ros-simulation/gazebo_ros_pkgs #1079: Configurable QoS settings for ROS 2 publishers and subscriptions

jacobperron, 2020-04-15:
## Description

Currently, all of the QoS settings for ROS 2 publishers and subscriptions created in Gazebo plugins are hardcoded (ie. cannot be changed by the user). In particular, many sensor plugins use the "SensorDataQoS" profile introduced in https://github.com/ros-simulation/gazebo_ros_pkgs/pull/926. This QoS profile is generally good for sensor data, using *best effort* for the reliability setting. But, there are cases where applications would like to ensure the arrival of sensor data and use *reliable* reliability.

IMO, gazebo_ros_pkgs should allow for the QoS settings to be entirely configurable so that users can mimic their real ROS 2 drivers accurately (e.g. maybe they have a camera driver that uses the *reliable* setting).

## Implementation considerations

ROS 2 does not officially support a way to configure QoS settings at runtime. Therefore, it is up to the node developer to enable this<sup>1</sup>. In any case, gazebo_ros_pkgs plugins and their parameters are already defined in SDF, so I think it makes sense to add SDF tags for setting various QoS settings for ROS 2 entities in a plugin. I propose introducing a new helper class for parsing QoS SDF tags and returning a `rclcpp::QoS` that can be used by the plugins.

I'm willing to do the work, but I'm looking for some feedback before I go ahead.

---
<sup>1</sup> I think there should be a standard mechanism builtin to ROS 2 to opt-in to configurable QoS settings as a node developer, but this is a separate discussion.

chapulina, 2020-04-15:
> I think there should be a standard mechanism builtin to ROS 2

+1, do you know if there's an issue ticketed for this, it would be nice to link it here.

> I think it makes sense to add SDF tags for setting various QoS settings for ROS 2 entities in a plugin

Sounds reasonable to me :+1: 

It may be worth it writing down the desired tags before implementing, i.e. `<ros><qos>`... If there's a plan to expose those standardly in ROS 2, as command line args for example, it would be nice to match the terminology so we can migrate more cleanly later.

wjwwood, 2020-04-15:
Seems like the right thing to do. If you're emulating some sensor and can configure things like rate or noise settings via SDF parameters, it makes sense to also be able to control how you emulate the communication settings.

jacobperron, 2020-04-15:
> If there's a plan to expose those standardly in ROS 2, as command line args for example, it would be nice to match the terminology so we can migrate more cleanly later

At the moment, there's not even a proposal written down, but I'd like to open up a design doc PR at some point. But hopefully the terminology won't be too different if we stick to DDS QoS terms.

ivanpauno, 2020-04-22:
This proposal sounds good to me, but I think we should have a more general solution.

For example, it makes sense for nodes in [image_pipeline](https://github.com/ros-perception/image_pipeline/) that different people will like to use them with different QoS settings for their applications.
In that case, how will the configuration be done? Exposing qos policies in parameters or command lines arguments will be extremely verbose IMO.

I think that a general approach could be to have a "QoS profiles file", but not exactly as it works in DDS.
That file could be pointed with an environment variable or maybe with a command line argument too.

About how the qos profile will be identified in the file for a specific publisher/subscription, I think it could work as follow:
- Subscription/Publisher qos profiles should be independent in the file.
- QoS profiles are specified for each topic name and topic type (topic type can be optionally specified).
- In case several nodes wants to create publishers/subscriptions in the same topic with different qos profiles, the file should also allow to specify a profile for a fully-qualified node name/topic name/topic type combination. That would override the qos profile specified for a topic in the file (that doesn't specify a node).

In the case the same node wants to create more than a publisher/subscription with different qos profiles in the same topic (which is an extreme corner case), I have several ideas, but that probably should go to a design doc if we want to consider this approach.

There should also be a way in the code to forbid the specified qos profile from being externally overridden.

ivanpauno, 2020-04-22:
To clarify, a qos profile file would look like:

```xml
<qos_profiles>
  <topic name="/my_topic" type="my_pkg/MyType">
    <publisher>
      ...QoS profile definition here
    </publisher>
    <subscription>
      ...
    </subscription>
  </topic>
  <node ns="/" name="my_node">
    <topic name="/my_topic" type="my_pkg/MyType">
      <publisher>
        ...QoS profile definition here
      </publisher>
      <subscription>
        ...
      </subscription>
    </topic>
  </node>
</qos_profiles>
```

For the node `/my_node`, the last qos profiles specified will be used, and for the others the ones at the beginning.

jacobperron, 2020-04-22:
> There should also be a way in the code to forbid the specified qos profile from being externally overridden.

I would argue that overriding externally should be disabled by default. I think having a default where node authors have to consider many possible QoS settings will lead to negligence or boilerplate code to disable a lot of options. I think it makes more sense for node authors to opt-in to certain configurations. E.g. topic X is allowed to have reliability "best effort" or "reliable". The choice could still be made with an external profile file, and a runtime error raised if an invalid option is chosen.

wjwwood, 2020-04-22:
Came here to say this ^

I strongly believe it should be opt in to support external override of QoS.

jacobperron, 2020-04-22:
I've opened a ticket here where we can move discussion about a more general ROS 2 approach: https://github.com/ros2/design/issues/280

ivanpauno, 2020-04-22:
> I would argue that overriding externally should be disabled by default

That makes a lot of sense to me.

> I think it makes more sense for node authors to opt-in to certain configurations. E.g. topic X is allowed to have reliability "best effort" or "reliable". The choice could still be made with an external profile file, and a runtime error raised if an invalid option is chosen.

For nodes of "general use" (i.e.: intended to be reused in different applications), I think it's easier to expose everything than trying to guess the different use-cases people will find.
I prefer trusting that developers are responsible adults that understand that overriding a qos policy will have a strong impact in the behavior of the application, than partially exposing them and then expose a new policy when someone comes up with a different use-case.
Opting-in the ability to externally configure a qos profile completely will also be much less verbose than explicitly exposing some policies.

TL;DR

I think a publisher/subscription should either allow the qos profile to be completely overridden or not, without intermediate alternatives. The default should be to not allow overrides.

jacobperron, 2020-06-12:
This feature was added in #1091 and #1092.

closed 2020-06-12 by jacobperron