ros-simulation/gazebo_ros_pkgs #1080: Release for ROS2 Foxy

ruffsl, 2020-04-16:
Opening a ticket to track the release of `gazebo_ros_pkgs` for ROS2 Foxy.

ruffsl, 2020-04-16:
`gazebo_ros_pkgs` is being used for a number of tests in `navigation2` packages, so releasing `gazebo_ros_pkgs` for the next ROS2 LTS and gazebo version supported with Ubuntu 20.04 would help in releasing the nav stack as well. https://github.com/ros-planning/navigation2/issues/1627

From the comments in the long standing ROS2 porting ticket, it seems that the ROS2 branch mabey out of sync with features and bug fixes from dashing and eloquent:  https://github.com/ros-simulation/gazebo_ros_pkgs/issues/512#issuecomment-588017483

chapulina, 2020-04-16:
[Here](http://repo.ros2.org/status_page/blocked_releases_foxy.html?q=id%3D%22gazebo_ros_pkgs%22&s=6&r=1)'s where to keep track of what packages are blocking us, there are currently 13 blocking packages upstream.

ruffsl, 2020-04-29:
Those packages seem to block a release sync, but what about a source build?
`gazebo_dev` still needs to be migrated to Gazebo 11 for Ubuntu 20.04, no?

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/00d0d67309039c21e92eaa65735347ba48598149/gazebo_dev/package.xml#L23-L24

chapulina, 2020-04-29:
This is needed to compile `vision_opencv` from source on Focal: https://github.com/ros-perception/vision_opencv/pull/324

SteveMacenski, 2020-05-27:
I pinged maintainers on vision_opencv and image_common to release so this can be released (and then navigation2 can be released). We should really start a new-distro-maintainers-release party post-COVID-19. Try to get as many out in a day as possible.

ruffsl, 2020-06-09:
I think `vision_opencv` and `image_common` have been released into foxy.
So there doesn't seem to be anything left blocking this.

doisyg, 2020-06-11:
Both `vision_opencv` and `image_common` have been released now, is there anything blocking a Foxy release ?

chapulina, 2020-06-11:
We started triaging the PRs targeted at the `ros2` branch to make sure there isn't anything we need to merge before the release:

https://github.com/ros-simulation/gazebo_ros_pkgs/pulls?q=is%3Apr+is%3Aopen+base%3Aros2

Someone also needs to forward-port the latest `dashing` and `eloquent` changes to `ros2`.

I also think we have many tests failing on the `ros2` branch right now, so it would be good to take a look at those before the release.

I won't have much time in the coming days to put into this, so help is appreciated.

- [ ] Review / merge / close [PRs targeting Foxy](https://github.com/ros-simulation/gazebo_ros_pkgs/pulls?q=is%3Apr+is%3Aopen+base%3Aros2)
- [ ] Forward-port `dashing` :arrow_right:  `eloquent` :arrow_right:  `ros2`
- [ ] Check / fix failing tests

chapulina, 2020-06-20:
All ready for a release, I just need an approval on https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1130

chapulina, 2020-06-20:
Phew! :smile: 

https://github.com/ros/rosdistro/pull/25542

closed 2020-06-20 by chapulina