ros-simulation/gazebo_ros_pkgs #1085: Release gazebo_ros_pkgs into ROS Noetic?

sloretz, 2020-04-23:
Would you mind releasing `gazebo_ros_pkgs` to ROS Noetic? It looks like all of its dependencies have been released, and recursively 62 are blocked by it.

chapulina, 2020-04-23:
I believe #1069 is needed.

@sloretz , is there already a way to run CI for Noetic?

sloretz, 2020-04-24:
> @sloretz , is there already a way to run CI for Noetic?

Yup. [It looks like this repo already gave the buildfarm permission to do PR jobs](http://wiki.ros.org/buildfarm/Pull%20request%20testing), so all that's needed is a source entry in the `noetic/distribution.yaml` file in `ros/rosdistro`. Here's an example the `ros_controllers` maintainer did: https://github.com/ros/rosdistro/pull/24553

seanyen, 2020-04-28:
Sorry to hijack the thread. Also hope this can be in for the `noetic` release: https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1089

sloretz, 2020-05-05:
@chapulina @j-rivero friendly ping

chapulina, 2020-05-06:
@sloretz , we need this: https://github.com/ros/rosdistro/pull/24646

sloretz, 2020-05-15:
Here are some recommendations for what to do with PRs for Noetic. This list is only as exhaustive as Github's search for `noetic`

* https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1100 - I think this was a proposal to support using ogre 1.12, but since ogre 1.9 is being used I think this one can be closed
* https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1096 - Small, already approved by @ahcorde ; I think this one can be merged
* https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1081 - already approved by @jacobperron ; I think this one can be merged
* https://github.com/ros-simulation/gazebo_ros_pkgs/pull/938 - small reviewed by @chapulina but not explicitly approved; I think this one can get a quick review and then be merged
* https://github.com/ros-simulation/gazebo_ros_pkgs/pull/902 - opened by me, it's ready to be reviewed and merged; tests pass locally
* https://github.com/ros-simulation/gazebo_ros_pkgs/pull/862 - looks like it's had a review in the past, but no explicit approval. needs a review then it can probably be merged

chapulina, 2020-05-15:
Thanks for triaging, @sloretz . I reviewed / merged / commented.

j-rivero, 2020-05-19:
First round of Noetic packages is ready, version 2.9.0. Testing and feedback is very welcome.

closed 2020-06-18 by chapulina