ros-simulation/gazebo_ros_pkgs #1088: Release ros-melodic-gazebo11-ros-pkgs / ros-dashing-gazebo11-ros-pkgs

chapulina, 2020-04-28:
Gazebo 11 will be the official version supported for ROS Noetic and Foxy. In order to ease the migration from earlier versions, it would be good to release `gazebo_ros_pkgs` for Gazebo 11 + earlier ROS LTS versions.

As a reference, these special package combinations are hosted [here](http://packages.osrfoundation.org/gazebo/ubuntu-stable/pool/main/r/) at http://packages.osrfoundation.org . [Here](http://gazebosim.org/tutorials?tut=ros_wrapper_versions&cat=connect_ros#UsingaspecificGazeboversionwithROS)'s a tutorial on how to use them, and [here](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/melodic-devel/docs/RELEASE.md)'s the documentation for the process of releasing these packages.

j-rivero, 2020-05-01:
@chapulina do we want dashing for any specific reason? Without more information Eloquent seems to me like the best candidate (latest release).

chapulina, 2020-05-01:
I suggested dashing because it's an LTS and it will outlive eloquent.

But in any case I think we should have both if possible.

moriarty, 2020-05-01:
There seems to be a `ros-melodic-gazebo-dev` on packages.orsf.org which depends on `libgazebo11-dev` instead of `libgazebo9-dev` should the `ros-melodic-gazebo-dev` package be `ros-melodic-gazeb11-dev` and the non versioned `ros-melodic-gazebo-dev` point to the `gazebo9` version?

j-rivero, 2020-05-01:
> @j-rivero thanks for the fix.

sorry for the problem :)

closed 2020-05-13 by j-rivero