ros-simulation/gazebo_ros_pkgs #1105: Gzclient not closing with sigint

c-rizz, 2020-05-14:
ROS melodic
Ubuntu 18.04

Gzclient does not close properly when it receives a SIGINT if launched through gazebo_ros.
When gazebo is started via gazebo_ros, gzclient does not cloesswith the SIGINT signal, but only when it receives SIGTERM.

This can be seen launching

```
roslaunch gazebo_ros empty_world.launch
```

I looked a bit into it and it seems it gets stuck here: [https://github.com/ros-simulation/gazebo_ros_pkgs/blob/f7a19b82f449044c5d39fc1f02f2fbb4471a7457/gazebo_ros/src/gazebo_ros_api_plugin.cpp#L88](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/f7a19b82f449044c5d39fc1f02f2fbb4471a7457/gazebo_ros/src/gazebo_ros_api_plugin.cpp#L88)

It gets stuck because the thread is caught in an infinite wait, it waits for  two services that don't exist for gazebo_gui.

Actually I don't really understand why the gazebo_ros_api plugin is being loaded in gazebo_gui. Is this intended?
Anyway, I tried disabling it by commenting lines 21-24 here: [https://github.com/ros-simulation/gazebo_ros_pkgs/blob/f7a19b82f449044c5d39fc1f02f2fbb4471a7457/gazebo_ros/scripts/gzclient#L21](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/f7a19b82f449044c5d39fc1f02f2fbb4471a7457/gazebo_ros/scripts/gzclient#L21)
It seems to solve the issue

If the solution makes sense to you I can make a pull request.

Also, I'm not sure if issue #751 is related, it seems to refer only to gzserver.

Thanks!

**Environment (please complete the following information):** 

 - OS: Ubuntu 18.04
 - Gazebo version: Gazebo 9
 - ROS version: Melodic
 - gazebo_ros_pkgs:  apt install, 2.8.6
 - glx server in use: server glx vendor string: NVIDIA Corporation

chapulina, 2020-05-14:
Thank you for diagnosing the problem.

> Actually I don't really understand why the gazebo_ros_api plugin is being loaded in gazebo_gui. Is this intended?

This commit message by @j-rivero  gives more context: https://github.com/ros-simulation/gazebo_ros_pkgs/commit/6083ed44d6a853d61fa7fe44292f85f95751fda2

Maybe there's a different solution that's more appropriate than loading that plugin.

c-rizz, 2020-05-14:
Hi, no problem
I also just noticed that the problem can alternatively be fixed by setting /gazebo_gui/enable_ros_network to false in empty_world.launch