ros-simulation/gazebo_ros_pkgs #1109: Could not find joint 'joint_1' in 'hardware_interface::PositionJointInterface'.

JunWeston, 2020-05-27:
hi, i am getting this error:

[ERROR] [1590570437.185619034, 0.556000000]: Could not find joint 'joint_1' in 'hardware_interface::PositionJointInterface'.
[ERROR] [1590570437.185702323, 0.556000000]: Failed to initialize the controller
[ERROR] [1590570437.185736701, 0.556000000]: Initializing controller 'fanuc_arm_controller' failed
[ERROR] [1590570438.186585, 1.536000]: Failed to load fanuc_arm_controller

while launching a urdf in gazebo. I am new to ROS and have looked for a solution but couldn't find it.
I would be glad if someone can help me with this. 

the detail information as below:
[ INFO] [1590570433.871991393]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1590570433.874217045]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1590570434.047505228]: Stereo is NOT SUPPORTED
[ INFO] [1590570434.047576597]: OpenGl version: 3.1 (GLSL 1.4).
[ INFO] [1590570435.988491588]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1590570436.016334066, 0.005000000]: Physics dynamic reconfigure ready.
Error:   Material [red] has malformed color rgba values: Unable to parse component [${255/255}] to a double (while parsing a color value)
         at line 93 in /build/urdfdom-YMMa9X/urdfdom-1.0.0/urdf_parser/src/link.cpp
[spawn_urdf-3] process has finished cleanly
log file: /home/westom/.ros/log/bde388b4-9fcc-11ea-bc53-6045cb60e2ee/spawn_urdf-3*.log
[ INFO] [1590570436.844600295, 0.386000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1590570436.846840028, 0.386000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1590570436.900669112, 0.386000000]: Loading gazebo_ros_control plugin
[ INFO] [1590570436.901019676, 0.386000000]: Starting gazebo_ros_control plugin in namespace: /fanuc
[ INFO] [1590570436.901560636, 0.386000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[ INFO] [1590570437.016229661, 0.386000000]: Loaded gazebo_ros_control.
[INFO] [1590570437.096931, 0.467000]: wait_for_service(/fanuc/controller_manager/load_controller): finally were able to contact [rosrpc://westom-System-Product-Name:51369]
[INFO] [1590570437.098113, 0.468000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1590570437.100328, 0.470000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1590570437.102764, 0.473000]: Loading controller: joint_state_controller
[INFO] [1590570437.110283, 0.481000]: Controller Spawner: Loaded controllers: joint_state_controller
[INFO] [1590570437.114060, 0.484000]: Started controllers: joint_state_controller
[INFO] [1590570437.167918, 0.537000]: wait_for_service(/fanuc/controller_manager/load_controller): finally were able to contact [rosrpc://westom-System-Product-Name:51369]
[INFO] [1590570437.168742, 0.538000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1590570437.170314, 0.540000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1590570437.171922, 0.541000]: Loading controller: fanuc_arm_controller

[ERROR] [1590570437.185619034, 0.556000000]: Could not find joint 'joint_1' in 'hardware_interface::PositionJointInterface'.
[ERROR] [1590570437.185702323, 0.556000000]: Failed to initialize the controller
[ERROR] [1590570437.185736701, 0.556000000]: Initializing controller 'fanuc_arm_controller' failed
[ERROR] [1590570438.186585, 1.536000]: Failed to load fanuc_arm_controller

launch and configs as below:

controller_gazebo.yaml
controller_list:
  - name: fanuc/fanuc_arm_controller
    action_ns: follow_joint_trajectory
    type: FollowJointTrajectory
    default: true
    joints:
      - joint_1
      - joint_2
      - joint_3
      - joint_4
      - joint_5
      - joint_6

fanuc_moveit_controller_manager.launch.xml
<launch>
  
  <arg name="moveit_controller_manager" default="moveit_simple_controller_manager/MoveItSimpleControllerManager" />
  <param name="moveit_controller_manager" value="$(arg moveit_controller_manager)"/>

  <arg name = "use_controller_manager" default="false"/>
  <param name = "use_controller_manager" value="$(arg use_controller_manager)"/>
  <param name="trajectory_execution/execution_duration_monitoring" value="false"/>

  
[... 107 lines cut ...]
        <type>transmission_interface/SimpleTransmission</type>
        <joint name="joint_5">
            <hardwareInterface>hardware_interface/PositionJointInterface</hardwareInterface>
        </joint>
        <actuator name="joint_5_motor">
            <hardwareInterface>hardware_interface/PositionJointInterface</hardwareInterface>
            <mechanicalReduction>1</mechanicalReduction>
        </actuator>
    </transmission>
    <transmission name="trans_joint_6">
        <type>transmission_interface/SimpleTransmission</type>
        <joint name="joint_6">
            <hardwareInterface>hardware_interface/PositionJointInterface</hardwareInterface>
        </joint>
        <actuator name="joint_6_motor">
            <hardwareInterface>hardware_interface/PositionJointInterface</hardwareInterface>
            <mechanicalReduction>1</mechanicalReduction>
        </actuator>
    </transmission>

    <gazebo>
        <plugin name="gazebo_ros_control" filename="libgazebo_ros_control.so">
            <robotNamespace>/fanuc</robotNamespace>
        </plugin>
    </gazebo>

JunWeston, 2020-05-27:
fanuc_moveit_controller_manager.launch.xml
<launch>
  
  <arg name="moveit_controller_manager" default="moveit_simple_controller_manager/MoveItSimpleControllerManager" />
  <param name="moveit_controller_manager" value="$(arg moveit_controller_manager)"/>

  <arg name = "use_controller_manager" default="false"/>
  <param name = "use_controller_manager" value="$(arg use_controller_manager)"/>
  <param name="trajectory_execution/execution_duration_monitoring" value="false"/>

  
  <rosparam file="$(find fanuc_moveit_config)/config/ros_controllers.yaml"/>
</launch>

fanuc_gazebo_joint_states.yaml
fanuc:
  joint_state_controller:
    type: joint_state_controller/JointStateController
    publish_rate: 50

fanuc_trajectory_control.yaml
fanuc:
  fanuc_arm_controller:
    type: "position_controllers/JointTrajectoryController"
    joints:
      - joint_1
      - joint_2
      - joint_3
      - joint_4
      - joint_5
      - joint_6

    gains:
      joint_1:   {p: 1000.0, i: 0.0, d: 0.1, i_clamp: 0.0}
      joint_2:   {p: 1000.0, i: 0.0, d: 0.1, i_clamp: 0.0}
      joint_3:   {p: 1000.0, i: 0.0, d: 0.1, i_clamp: 0.0}
      joint_4:   {p: 1000.0, i: 0.0, d: 0.1, i_clamp: 0.0}
      joint_5:   {p: 1000.0, i: 0.0, d: 0.1, i_clamp: 0.0}
      joint_6:   {p: 1000.0, i: 0.0, d: 0.1, i_clamp: 0.0}  

fanuc_gazebo_states.launch
<launch>
    <rosparam file="$(find fanuc_gazebo)/config/fanuc_gazebo_joint_states.yaml" command="load"/>
    
    <node name="joint_controller_spawner" pkg="controller_manager" type="spawner" respawn="false"
          output="screen" ns="/fanuc" args="joint_state_controller" />
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher"
        respawn="false" output="screen">
        <remap from="/joint_states" to="/fanuc/joint_states" />
    </node>
</launch>

fanuc_gazebo_world.launch
<launch>
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find fanuc_gazebo)/world/fanuc.world"/>
  </include>
  <node name="spawn_urdf" pkg="gazebo_ros" type="spawn_model" args="-file $(find fanuc_description)/urdf/fanuc.urdf -urdf -x 0 -y 0 -z 1 -model fanuc" />
</launch>

fanuc_trajectory_controller.launch
<launch>
    <rosparam file="$(find fanuc_gazebo)/config/fanuc_trajectory_control.yaml" command="load"/>
    <node name="arm_controller_spawner" pkg="controller_manager" type="spawner" respawn="false"
          output="screen" ns="/fanuc" args="fanuc_arm_controller"/>
</launch>

fanuc_bringup_moveit.launch
<launch>
    <include file="$(find fanuc_gazebo)/launch/fanuc_gazebo_world.launch" />
    <include file="$(find fanuc_gazebo)/launch/fanuc_gazebo_states.launch" />  
    <include file="$(find fanuc_gazebo)/launch/fanuc_trajectory_controller.launch" />
    <include file="$(find fanuc_moveit_config)/launch/moveit_planning_execution.launch" />
</launch>

JunWeston, 2020-05-27:
fanuc_moveit_controller_manager.launch.xml
<launch>
  
  <arg name="moveit_controller_manager" default="moveit_simple_controller_manager/MoveItSimpleControllerManager" />
  <param name="moveit_controller_manager" value="$(arg moveit_controller_manager)"/>

chapulina, 2020-05-27:
>  I am new to ROS

Welcome! :v: 

> I would be glad if someone can help me with this.

Please ask for help at http://answers.gazebosim.org/ . This issue tracker is for bug reports. I'm going to close this. If you believe you've encountered a bug, please reopen this issue explaining why you think it's a bug.

(also, a quick tip, always format your code into code blocks like I did it for you above, so it's easier to read)

closed 2020-05-27 by chapulina

JunWeston, 2020-05-28:
@chapulina thanks!