ros-simulation/gazebo_ros_pkgs #1112: libgazebo_ros_factory.so cannot find models

rjshim, 2020-05-29:
Describe your issue here and explain how to reproduce it.

1. Prepare your urdf file with <mesh filename="package://~"> lines
2. gazebo --verbose -s libgazebo_ros_factory.so
3. ros2 run gazebo_ros spawn_entity.py -entity myentity -x 0 -y 0 -z 0 -file ~/path_to_urdf_file/myrobot.urdf
Then you will see error codes with "Unable to download model[model://~]". 
I didn't see this problem when I read my urdf file and displayed on rviz2 but only when I tried to spawn it on Gazebo.

A clear and concise description of what the bug or feature is.

Read urdf file with <mesh filename="package://~"> and spawn the robot model on gazebo does not work with libgazebo_ros_factory.so.

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**

A clear and concise description of what you expected to happen.

**Screenshots**

If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

 - OS: Ubuntu 18.04
 - Gazebo version:  Gazebo 9
 - ROS version: Dashing
  - gazebo_ros_pkgs: debian install
 - glx server in use: output from `glxinfo | grep 'server glx vendor'` (glxinfo is in mesa-utils package for Debian/Ubuntu)

**Additional context**

Add any other context here.

joe28965, 2020-06-10:
So this has to do with the difference of setup between ROS1 and ROS2. Gazebo will look in the paths it has available for the files. You basically need to tell it where to find it. In the link below it explains how to do just that.

https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-Gazebo-ROS-Paths

Just to give you a longer explanation, you locate where your meshes are. Let's assume your package looks something like this:

```
my_package
   meshes
      my_mesh.stl
   package.xml
```

In your package, you need to add the following:

```
  <export>
    <build_type>ament_cmake</build_type>
    <gazebo_ros gazebo_model_path="${prefix}/meshes"/>
  </export>
```

The `${prefix}` points towards the prefix of package.xml, so by adding `/meshes` do you point towards the path of your mesh. When you run the entity spawner now, it should be able to find your file.

If this solves your problem, please let us know so we can close the issue.

rjshim, 2020-06-11:
@joe28965 

It still does not work and now I have two questions.
1. If I directly give the gazebo_model_path by typing "export GAZEBO_MODEL_PATH=~" in the terminal window, it says that I have to add model.config files, which is not written on the [instruction](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-Gazebo-ROS-Paths). Is it necessary to add model.config files? 
2. If there are any good implementation examples of the instruction, it will be really helpful. I did as your advice and followed this [instruction](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-Gazebo-ROS-Paths) but gazebo still says it cannot find the path.

Thank you for help in advance.

chapulina, 2020-06-11:
@rjshim , there's a large number of things that could be missing / wrong in your setup. I think it will be difficult to help unless you provide a minimal example of the issue you're facing.

joe28965, 2020-06-11:
@rjshim 
For point 1, there is a difference between `export GAZEBO_MODEL_PATH=~` in the terminal and `<gazebo_ros gazebo_model_path="${prefix}/models"/>` in your package.xml file. The first one is a gazebo thing, the second one a gazebo_ros thing. For the second option you do not need to add a model.config file.
Just to clarify, when I said in my first message to add this to your package:
```
  <export>
    <build_type>ament_cmake</build_type>
    <gazebo_ros gazebo_model_path="${prefix}/meshes"/>
  </export>
```
I meant adding it to package.xml.

For point 2, I do not know, the stuff I work on is not public. But if you give us some more information as chapulina has requested, we could take a look for you.

Finally, the one thing I am wondering though, is how you start Gazebo. Because the launch files provided by gazebo_ros:
- [gzserver](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/36d7a9e7242fe244506d64acb373f8450a19461a/gazebo_ros/launch/gzserver.launch.py#L30)
- [gzclient](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/36d7a9e7242fe244506d64acb373f8450a19461a/gazebo_ros/launch/gzclient.launch.py#L30)

you can see that in line 30-41 it grabs the `gazebo_ros gazebo_model_path` that you set in your package.xml file and converts them to `GAZEBO_MODEL_PATH` for Gazebo. (If I understand that part correcty @chapulina ?) If you start gazebo yourself and you do not include this, that could cause issues for you.

chapulina, 2020-06-11:
> If I understand that part correcty

Yes, that's right. There's also the alternative using [env hooks](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-Gazebo-ROS-Paths#env-hooks) also explained on the tutorial, which should be more general and work even without those launch files.

rjshim, 2020-06-23:
@joe28965 @chapulina 

Thank you very much for your reply. I read your comments over and over and over. I also tested this on different machines & OS (Ubuntu 18.04, 20.04) & ROS versions (Dashing, Foxy). While testing it, I could get a deeper understanding of how Gazebo and gazebo_ros work together but still there are several points that I haven't figured out yet. Let me rephrase your answers and clarify my questions below. 

1) There are three ways you can pass GAZEBO_MODEL_PATH to Gazebo. 
- Directly "export GAZEBO_MODEL_PATH = ~~" in the terminal window. 
- Export <gazebo_ros gazebo_model_path="${prefix}/meshes"/> in your package.xml file and build. 
- Env hooks. 

I was mainly interested in the second method since it is the most intuitive and easy way of porting ROS1 codes to ROS2. To use it, you should run gzserver or gzclient with Gazebo together since they will collect those newly exported paths. But if you read the code that @joe28965 linked, you can obviously see that if you don't have GAZEBO_MODEL_PATH already in environ, gzserver and gzclient wouldn't add new exported paths. In other words, you can use the second method only when you are already using the first method for another model path. If there is no output from typing "echo $GAZEBO_MODEL_PATH" in the terminal window, the second method wouldn't work.

2) You might still need model.config files for the second method.
@joe28965 said that I do not need model.config files for the second method, but it still does not work for me without them. To know if it is because I tried to make Gazebo find models without model.config files, I used the second method and tested if Gazebo can find models with model.config files, and this time, it worked fine without any issues. It kind of makes sense as Gazebo still need model.config files no matter which methods you use to pass model paths to it.

But in ROS1, we didn't have to add model.config files. We could simply register our urdf files in the param server and spawn them using the 'spawn_model' executable in the 'gazebo_ros' package. In ROS2, it seems adding model.config files to each model file is necessary. (Let me know if I am misunderstanding something)

Since what I am working on is also not public, I asked if there is any sample that I can refer to. Honestly, digging in Gazebo and gazebo_ros packages is not really easy without a grasp of how they are structured. If there exists, any sample examples will be really helpful. I also checked the turtlebot3 gazebo package written in ROS2, but it also uses model.config files.

Any advice and suggestions will be appreciated. Thank you.
Ryan

rjshim, 2020-06-24:
Thank you very much for your answers. 

Using <gazebo_ros gazebo_model_path="${prefix}/../"/> instead of <gazebo_ros gazebo_model_path="${prefix}/meshes"/> made everything work. 
Considering urdf files will be also used by robot_state_publisher, unifying model paths in the urdf files will be necessary and then, gazebo_model_path should also be set considering it. 

Thank you very much for your help and I will close this issue.
Ryan

closed 2020-06-24 by rjshim

DentOpt, 2021-01-22:
Hi guys, 
I have a similar problem with the mesh linking. 

```
my_ros2_robot_description_package/
-robots/
--robot1/
---robot1.urdf
---meshes/
-----mesh1.STL
CMakeLists.txt
package.xml
```
in my robot1.urdf file I have the reference to 
<mesh filename="package://my_ros2_robot_description_package/robots/robot1/meshes/mesh1.STL" />

If I launch the robot description:
`ros2 run robot_state_publisher robot_state_publisher staark_4dof_prototype.urdf`  
I can visualize the robot nicely in rviz2. 

However, if I then try to import the model in gazebo based on the ros2 robot_description topic, it does not find the stl:
`ros2 run gazebo_ros spawn_entity.py -entity model -topic /robot_description`  
This is the error:
` Failed to find mesh file [model://my_ros2_robot_description_package/robots/robot1/meshes/mesh1.STL]`

I tried to define the model path in the package.xml by adding 
```
  `<depend>gazebo_ros</depend> <!-- needed for relative mesh paths for gazebo-->`

  <export>
    <build_type>ament_cmake</build_type>
    <gazebo_ros gazebo_model_path="${prefix}/robots/robot1/meshes"/>  <!-- needed for relative mesh paths for gazebo-->
  </export>
```
I also tried different paths with similar results
```
<gazebo_ros gazebo_model_path="${prefix}/../"/>
<gazebo_ros gazebo_model_path="${prefix}/"/>
<gazebo_ros gazebo_model_path="${prefix}/robots/"/>
```

Do you know where can I check which absolute gazebo_model_path is exported (e.g. `$GAZEBO_MODEL_PATH`)? 
Do you know what the entry point for the model is?

Thanks  a lot!

joe28965, 2021-01-22:
@DentOpt in order to use the gazebo_model_path in gazebo, you will need to launch it using the provided gazebo_ros launch files.
I mentioned this in my second reply to this thread

> 
> Finally, the one thing I am wondering though, is how you start Gazebo. Because the launch files provided by gazebo_ros:
> 
>     gzserver
>     gzclient
> 
> you can see that in line 30-41 it grabs the gazebo_ros gazebo_model_path that you set in your package.xml file and converts them to GAZEBO_MODEL_PATH for Gazebo. If you start gazebo yourself and you do not include this, that could cause issues for you.

However, I have since stopped using it this way, and started using [env-hooks](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-Gazebo-ROS-Paths#env-hooks). I find these to be a lot less of a hassle, and they do not require you to use the gzserver launch file of gazebo_ros.

joe28965, 2021-01-22:
@DentOpt I have added an example to the wiki on [env-hooks](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-Gazebo-ROS-Paths#env-hooks).

@chapulina I hope it's OK that I added the example. I've run into multiple people having issues with the model paths, and since I personally find env-hooks to be the cleanest of solutions, I thought an example might be very handy for people. If you have any feedback, please let me know.

DentOpt, 2021-01-22:
Perfect, thanks a lot that helped! 
It's working now using env-hooks!

Regarding cleanliness:
If you want to use the same urdf for gazebo and rviz this becomes a bit dirty. 
Rviz visualization uses the /share as root, e.g. 
       ` <package://my_ros2_robot_description_package/robots/robot1/meshes/mesh1.STL" />`  
If I use 
` prepend-non-duplicate;GAZEBO_MODEL_PATH;share` as hook the urdf can find the meshes, 
but not the model.config file. 

If I use  
` prepend-non-duplicate;GAZEBO_MODEL_PATH;share/my_ros2_robot_description_package/robots/`   
to the model folder and ` <package://robot1/meshes/mesh1.STL" />`  that works smoothly for Gazebo, 
but of course in RVIZ the urdf does not relate to the proper mesh path anymore. 

Is there a smart way to consolidate this?

```
-robots/
--robot1/
---model.config
```

joe28965, 2021-01-22:
Gazebo doesn't understand the `package://` prefix, and rviz doesn't understand the `model://` prefix.

In the turtlebot package, they have just simply made 2 models, one for gazebo and one for rviz.

In my own package, I use xacro for my project, and have this structure:

```
robot_description_package/
-env-hooks/
--robot_description_package.dsv.in
-models
--my_robot_model/
---meshes/
----my_mesh.stl
---my_robot.urdf
---model.config
-urdf/
--my_robot.urdf.xacro
CMakeLists.txt
package.xml
```

And then in my xacro file I have the following 2 lines:

```
<xacro:property name="full_path_to_meshes" value="file://$(find robot_description_package)/models/my_robot_model/meshes"/>

...

  <visual>
    <geometry>
      <mesh filename="${full_path_to_meshes}/my_mesh.stl"/>
    </geometry>
  </visual>
```

When I run the xacro command to make my urdf file (and have it placed in the `my_robot_model` folder) it has the entire path in there. It's a very ugly solution, but I have not found a way to either teach rviz the `model://` prefix, or gazebo the `package://` one.

DentOpt, 2021-02-07:
Thank you for your support! I solved the issue for gazebo and rviz using one urdf by adding two hooks -> One for the project install repository and one to the model file repo. This is a bit dirty, as gazebo is searching model.config files in the project install repo on start up. However, it solves the depency issue.