ros-simulation/gazebo_ros_pkgs #1116: [ROS 2] gazebo_ros returns always an empty namespace

ahcorde, 2020-06-08:
I was trying to add a namespace using the `node.cpp` class. In particular I have some issue with this [line](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/ros2/gazebo_ros/src/node.cpp#L58). if I change the name of the xml tag to `name_space` or others I can get the value, but using the word namespace the code is returning always "/".

Arguments, parameters and remapping are working fine

```
<plugin ...>
            <ros>
              <namespace>gazebo_ros2_control</namespace>
            </ros>
</plugin>
```

@chapulina any ideas?

ahcorde, 2020-06-08:
This is the `sdf->PrintValues()` in the first line of the `Load` function (entry point for the plugin).For some reason it's overriding the `<namespace>` tag inside `<ros>`

[gzserver-1] <plugin name='gazebo_ros2_control' filename='libgazebo_ros2_control.so'>
[gzserver-1]   <ros>
[gzserver-1]     <namespace>/</namespace>
[gzserver-1]     <namespace2>test_cam_16bit</namespace2>
[gzserver-1]     <remapping>test_camera_name/image_raw:=image_test_16bit</remapping>
[gzserver-1]     <remapping>camera_info:=camera_info_test_16bit</remapping>
[gzserver-1]     <argument>from:=to</argument>
[gzserver-1]   </ros>

chapulina, 2020-06-08:
I also ran into this issue [yesterday](https://github.com/chapulina/dolly/commit/f9b2e078ce1b645be558708b3060a4cd4f1c57d7#diff-3ea051f5bf6c350c7601fe2d685f31c7R334) using `eloquent` + `gazebo9` debs. I'm sure it used to work. I have no clue why it would stop working now.

ahcorde, 2020-06-23:
Fixed in this PR https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1117

closed 2020-06-23 by ahcorde