ros-simulation/gazebo_ros_pkgs #1134: error in gazebo_ros_p3d.hpp

abramie, 2020-07-07:
Hi, I notice an arror in gazebo_ros_p3d.hpp, the exemple in top of the file use the wrong filename, it's <plugin name="gazebo_ros_force" filename="libgazebo_ros_force.so"> instead of <plugin name="gazebo_ros_force" filename="libgazebo_ros_p3d.so">

chapulina, 2020-07-07:
@abramie , it looks like your text got cut there. Could you add more context to this issue? Specific error messages and links to files are appreciated.

abramie, 2020-07-08:
[https://github.com/ros-simulation/gazebo_ros_pkgs/blob/dashing/gazebo_plugins/include/gazebo_plugins/gazebo_ros_p3d.hpp](url) in this file (forgot the issues aren't branch-related, and that I needed to check code for the XML to show up ^^ ).
Probably just a copy past mistake :)