ros-simulation/gazebo_ros_pkgs #1138: Cannot locate rosdep definition for [gazebo_ros_pkgs]

rjshim, 2020-07-23:
**Describe your issue here and explain how to reproduce it.
A clear and concise description of what the bug or feature is.**

1) Install foxy on Ubuntu 20.04.
2) make your own ros2 package
3) Add gazebo_ros_pkgs as one of the dependencies in the package.xml file.
4) install dependencies using rosdep then you will see the below error code
"Cannot locate rosdep definition for [gazebo_ros_pkgs]"
5) I can still install gazebo_ros_pkgs by typing below
$ sudo apt install ros-foxy-gazebo-ros-pkgs

**Expected behavior**
gazebo_ros_pkgs should be install also by rosdep. 
Is there a sort of rosdep server and every binary released package has to be registered there? 
I do not know how it is structured at the bottom layer of software, but if so, maybe it has to be registered. 

**Screenshots**

[image]

**Environment (please complete the following information):**

 - OS: Ubuntu 20.04
 - Gazebo version: Gazebo 11
 - ROS version: Foxy
  - gazebo_ros_pkgs: X
 - glx server in use: Not sure about this

**Additional context**

Thank you for your help in advance.
Ryan

chapulina, 2020-07-23:
Humm can you try `rosdep update`?

rjshim, 2020-07-23:
Thank you. It fixed the problem.

closed 2020-07-23 by rjshim