ros-simulation/gazebo_ros_pkgs #1158: Unable to locate package ros-foxy-gazebo-ros-pkgs

MarcoAndrade26, 2020-09-06:
I am trying to install gazebo-ros-pakgs running the following commands on Ubuntu 20.04:
sudo apt update
sudo apt upgrade
sudo apt install python3-colcon-common-extensions
sudo apt install gazebo11 libgazebo11-dev

But when running:
sudo apt install ros-foxy-gazebo-ros-pkgs

It shows me the error: E: Unable to locate package ros-foxy-gazebo-ros-pkgs

I tried also with:
sudo apt install gazebo11 ros-foxy-gazebo-ros-pkgs

But I am getting the same error.

**Environment:**

 - Board: Raspberry Pi4 4Gb of RAM
 - OS: Ubuntu 20.04
 - Gazebo version: Gazebo11
 - ROS version: Foxy
 - gazebo_ros_pkgs: Debian install

**Additional context**

Add any other context here.

j-rivero, 2020-09-07:
Have you installed ROS2 before trying to install the foxy packages? Could be that you don't have the ROS2 repositories available.

MarcoAndrade26, 2020-09-08:
I have already installed ROS2 and source it

j-rivero, 2020-09-08:
may I have your output from:

```
sudo apt-get update
```
and
```
apt-cache search ros-foxy-gazebo*
```

mrunaljsarvaiya, 2020-09-10:
I actually have the exact same issue. 
This is my output after running `apt-cache search ros-foxy-gazebo*`

`ros-foxy-gazebo-dev - Provides a cmake config for the default version of Gazebo for the ROS distribution.
ros-foxy-gazebo-msgs - Message and service data structures for interacting with Gazebo from ROS2.
ros-foxy-gazebo-msgs-dbgsym - debug symbols for ros-foxy-gazebo-m`

chapulina, 2020-09-10:
> Board: Raspberry Pi4 4Gb of RAM

> ros foxy on a Rpi4

This is the problem, `gazebo-ros-pkgs` is only available for amd64:

http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-foxy-gazebo-ros-pkgs/

MarcoAndrade26, 2020-09-10:
@chapulina Thanks! Does anyone know how to disable Gazebo when building Navigation2?

j-rivero, 2020-09-11:
> This is the problem, `gazebo-ros-pkgs` is only available for amd64:

Checking the root of the problem for this, I see _arm64 packages name in http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-foxy-gazebo-ros-pkgs/ dated on 09/04.

If I'm not wrong, this is the tracker for arm64 builds: http://repo.ros2.org/status_page/ros_foxy_ubv8.html?q=gazebo_ros_pkgs

samuk, 2022-08-05:
Predictably this is still an issue on Humble/ Pi4

j-rivero, 2022-08-05:
> Predictably this is still an issue on Humble/ Pi4

Not probably the same problem since platforms changed from Foxy to Humble. Humble is probably affected by https://github.com/osrf/gazebo/issues/3236, please follow the issue there.

Thanks.

closed 2022-08-05 by j-rivero