ros-simulation/gazebo_ros_pkgs #1163: Foxy CI Linux install failing

jacobperron, 2020-09-21:
**Description**

It looks like the CI fails with an assertion failure when trying run a demo world as part of CI:

```
[Err] [RenderEngine.cc:195] Failed to initialize scene
[Err] [GLWidget.cc:172] GLWidget could not create a scene. This will likely result in a blank screen.
gzclient: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::rendering::Camera; typename boost::detail::sp_member_access<T>::type = gazebo::rendering::Camera*]: Assertion `px != 0' failed. 
```

https://build.osrfoundation.org/job/ros2_gazebo_pkgs-install_pkg_foxy-focal-amd64/13

jacobperron, 2020-09-21:
I spent some time trying to reproduce with the Dockerfile and build.sh CI artifacts, but didn't figure out how to workaround a display issue running remotely in Docker.

jacobperron, 2020-09-22:
Looks like it may have been a flake as the build is back to normal: https://build.osrfoundation.org/job/ros2_gazebo_pkgs-install_pkg_foxy-focal-amd64/14/

I'll close this for now and keep an eye on it.

jacobperron, 2020-09-22:
Looks like the same failure for Eloquent CI: https://build.osrfoundation.org/job/ros2_gazebo11_pkgs-install_pkg_eloquent-bionic-amd64/128

jacobperron, 2020-09-22:
@j-rivero @chapulina FYI.

chapulina, 2020-09-22:
:monocle_face: I went through the past builds and it looks like the ones that failed with the rendering issue ran on r2d2. I triggered a new build, hopefully that's fixed now that the offending node no longer runs GPU jobs?

j-rivero, 2020-09-23:
last two CI runs are good on drogon. So the problem was probably the r2d2 node. Please reopen if the problem persist on other runs.

closed 2020-09-23 by j-rivero