ros-simulation/gazebo_ros_pkgs #1164: 404 on vital step in the install from source instructions

guru-florida, 2020-09-22:
Describe your issue here and explain how to reproduce it.

The [installation from source](http://gazebosim.org/tutorials?tut=ros2_installing&cat=connect_ros) (linked) page describes in step 3 to download a ros2.yaml file from bitbucket snippet (user chapulina?) and the snippet is no longer available. This 404 is in both the linked "Copy this file" and the wget command step:

Specific links are: 
1. (copy this) https://bitbucket.org/snippets/chapulina/geRKyA/ros2repos-supplement-gazebo_ros_pkgs
2. (wget) https://bitbucket.org/api/2.0/snippets/chapulina/geRKyA/f02dcd15c2c3b83b2d6aac00afe281162800da74/files/ros2.yaml

There is no alternative location I can find or determine the contents of this yaml file, and wayback machine didnt archive the above snippet unfortunately.

**Expected behavior**

With the right link the proper dependencies would be downloaded for gazebo_ros_pkgs.

Thank you.

scpeters, 2020-09-22:
cc @chapulina

jacobperron, 2020-09-22:
I think we should keep a version of the repos file in this repository (and it can be versioned per distro). See https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1166.

jacobperron, 2020-09-22:
Here's a PR updating the tutorial https://github.com/osrf/gazebo_tutorials/pull/120

j-rivero, 2020-09-23:
Thanks @guru-florida , @jacobperron , @chapulina . Instructions should be updated as per osrf/gazebo_tutorials#120

closed 2020-09-23 by j-rivero

guru-florida, 2020-09-23:
Awesome guys! Thanks.