ros-simulation/gazebo_ros_pkgs #1165: Foxy CI test failures "boost thread: trying joining itself: Resource deadlock avoided"

jacobperron, 2020-09-22:
Several tests are failing on Foxy CI due to a boost thread error:

https://build.osrfoundation.org/job/ros2_gazebo_pkgs-ci-default_foxy-focal-amd64/22

> terminate called after throwing an instance of 'boost::wrapexcept<boost::thread_resource_error>'
>   what():  boost thread: trying joining itself: Resource deadlock avoided

The following tests are failing in the referenced job:

- test_gazebo_ros_joint_effort
- test_gazebo_ros_diff_drive
- test_gazebo_ros_harness

jacobperron, 2020-09-23:
Another instance of the same error on Dashing CI: https://build.osrfoundation.org/job/ros2_gazebo11_pkgs-ci-default_dashing-bionic-amd64/6/

The following test failed:

- test_gazebo_ros_properties

jacobperron, 2020-09-23:
Again on Foxy CI during test_gazebo_ros_tricycle_drive: https://build.osrfoundation.org/job/ros2_gazebo_pkgs-ci-default_foxy-focal-amd64/23

scpeters, 2020-11-18:
I have seen these failures in upstream gazebo tests as well